compas-dev/compas #706: Coordinates are not set automatically in mesh.add_vertex(...)

isabelmdo, 2020-12-11:
**Describe the bug**
Coordinates are not set automatically.
When using `mesh.add_vertex(x=0, y=0, z=0)` we have to add all coordinates manually. If I pass in `mesh.add_vertex(x=1)`, y and z coordinates will not be created automatically as the [mesh tutorial](https://compas.dev/compas/latest/tutorial/meshes.html) suggests and geometry methods will fail.

**To Reproduce**
Steps to reproduce the behavior:
1. Context: file `mesh_example.py`
2. Sample script
```
from compas.datastructures import Mesh
mesh = Mesh()

a = mesh.add_vertex()  # x,y,z coordinates are optional and default to 0,0,0
b = mesh.add_vertex(x=1)
c = mesh.add_vertex(x=1, y=1)
d = mesh.add_vertex(y=1)

mesh.add_face([a, b, c, d])

print("mesh face: ", mesh.face)

print("topology method: ", mesh.vertex_neighbors(a, ordered=False))
print("geometry method: ", mesh.vertex_coordinates(a))
```

3. Output of `$ python mesh_example.py`:
```
mesh face:  {0: [0, 1, 2, 3]}
topology method:  [1, 3]
Traceback (most recent call last):
  File "mesh_example.py", line 14, in <module>
    print("geometry method: ", mesh.vertex_coordinates(a))
  File "C:\Users\(...)\mesh_example.py", line 814, in vertex_coordinates
    return [self.vertex[key][axis] for axis in axes]
  File "C:\Users\(...)\mesh_example.py", line 814, in <listcomp>
    return [self.vertex[key][axis] for axis in axes]
KeyError: 'x'
```

**Expected behavior**
Expected output to generate coordinates automatically.

**Screenshots**
Here is a screenshot of the [mesh tutorial example](https://compas.dev/compas/latest/tutorial/meshes.html):
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python version: 3.8.6
 - Python package manager: conda

**Additional context**
If this is the code snippet, then it works.
```
from compas.datastructures import Mesh
mesh = Mesh()

a = mesh.add_vertex(x=0, y=0, z=0)
b = mesh.add_vertex(x=1, y=0, z=0)
c = mesh.add_vertex(x=1, y=1, z=0)
d = mesh.add_vertex(x=0, y=1, z=0)

mesh.add_face([a, b, c, d])

print("mesh face: ", mesh.face)

print("topology method: ", mesh.vertex_neighbors(a, ordered=False))
print("geometry method: ", mesh.vertex_coordinates(a))
```
Output:
```
mesh face:  {0: [0, 1, 2, 3]}
topology method:  [1, 3]
geometry method:  [0, 0, 0]
```

brgcode, 2020-12-11:
ha interesting. the bug is actually related to the internal details of `mesh.vertex_coordinates`. if you do the following it works as advertised
```python
from compas.datastructures import Mesh

mesh = Mesh()

a = mesh.add_vertex()
b = mesh.add_vertex(x=1)

a_xyz = mesh.vertex_attributes(a, 'xyz')
b_xyz = mesh.vertex_attributes(b, 'xyz')
```

brgcode, 2020-12-11:
will fix

isabelmdo, 2020-12-11:
Ohh I see, thanks!

closed 2020-12-16 by brgcode

gonzalocasas, 2020-12-16:
@brgcode this would be a good candidate to add a unit test that verified expected behavior now that it's fixed.

brgcode, 2020-12-16:
done

commit 7022a042 (2020-12-16): closes #706