compas-dev/compas #710: `compas_rhino.utilities.delete_layers` deletes whole tree when given dict as input.

tetov, 2020-12-13:
## Describe the bug
Unexpected behavior when using a dict as parameter for `compas_rhino.utilities.delete_layer`.

(I might be misunderstanding though, let me know if so and I can probably updated docstring)

## To Reproduce

### Environment
Rhino Python Editor

### Sample script
```
from compas_rhino.utilities import create_layers, delete_layers
layers = {'COMPAS': {'layers': {'Datastructures': {'layers': {'Mesh': {}, 'Network': {}}}}}, "SAPMOC": {}}
create_layers(layers)
delete_layers({'COMPAS': {'layers': {'Datastructures': {'layers': {'Network': {}}}}}})
```

### Result

[image]

## Expected result

[image]

### Other environment details
**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python version: 2.7.8 (IronPython 2.7.8 (2.7.8.0) on .NET 4.0.30319.42000 (64-bit))
 - Python package manager: conda

brgcode, 2020-12-14:
perhaps we should provide the option to delete "endpoints" only?

tetov, 2020-12-14:
Maybe, but that can be done with `"COMPAS::Datastructures::Network"`.. I think just making sure that the more specific the input is, the more specific the action is. If the dict given contains sublayers down to an endpoint only the endpoint is removed. If it stops before then the last points tree is removed.