compas-dev/compas #723: Extremly poor performance when using compas_json.load() in Rhino

mattiskoh, 2020-12-23:
**Describe the bug**
Loading a small sized .json file (<1MB) in Rhino using compas_json.load() it takes many orders of magnitude longer then using the standard json library.

Using the compas version took me 20+ minutes, whereas the standard json.load() took less than a second.

**To Reproduce**
Steps to reproduce the behavior:
1) load a .json file using Rhino's Python editor 
2) run 
`from compas._json import json_load
filename = data_path + "fem_frames.json"
#frames = json_load(filename)`
3) wait for hours... 

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version 2.7

**Additional context**
Add any other context about the problem here.

brgcode, 2020-12-23:
can you share the file that is causing the problem?

brgcode, 2021-01-04:
ping

mattiskoh, 2021-01-05:
[Here is the json file](https://jsonblob.com/57b97b11-4f30-11eb-bace-712e25fd5735)

brgcode, 2021-01-11:
how do i download from there?

mattiskoh, 2021-01-25:
https://polybox.ethz.ch/index.php/s/eeuzXCM7kVKWgS9

Sorry for the late response github is not sending me notifications for this thread

BTW, the same problem also exists for json_dump performance

brgcode, 2021-01-25:
i can't really reproduce this.
could you run this script and let me know what the timings are on your machine?
FWIW, i get 
```
('json load', 0.76806640625)
('pickle dump', 3.329254150390625)
('pickle load', 1.68267822265625)
```

```python
import compas
import time
import os
import pickle

HERE = os.path.dirname(__file__)
JSON = os.path.join(HERE, 'test.json')
PICK = os.path.join(HERE, 'test.pick')

t0 = time.time()

data = compas.json_load(JSON)

t1 = time.time()

with open(PICK, 'wb') as f:
    pickle.dump(data, f)

t2 = time.time()

with open(PICK, 'rb') as f:
    data = pickle.load(f)

t3 = time.time()

print("json load", t1 - t0)
print("pickle dump", t2 - t1)
print("pickle load", t3 - t2)
```

brgcode, 2021-01-25:
just out of curiosity, in which context was this file produced? it has more than 69k lines and 3450 serialised objects (so i would not necessarily call it small, despite its size).

also, fyi, in Python 3, pickling is quite a bit faster...

brgcode, 2021-01-29:
@mattiskoh ping

mattiskoh, 2021-02-01:
Here are the results from the test script run from powerhsell: 
`json load 0.7793033123016357
pickle dump 0.10285520553588867`

These are the results from the test script run from Rhino Python:
`('json load', 0.8220672607421875)
('pickle dump', 1.6156158447265625)`

I noticed that there were some brackets misplaced in the .json file which blocked the script from completing. I manually fixed this just now.

The context; Of a larger mesh, I was trying to obtain the frames per mesh face (aligned to certain inputs), that I can then use to set the anisotropic material direction in FEM... 
I wanted to do a visual inspection in Rhino and there I noticed that really long for the script to complete.. 

Long story short, **I can't seem to reproduce the error anymore either..** 

so feel free to close.. Sorry for the delayed responses. really don't know why I get email notifications for other threads but not this one even though it says I'm subscribed...

closed 2021-02-01 by mattiskoh

brgcode, 2021-02-01:
no worries. all good. happy to hear the problem is resolved...