compas-dev/compas #728: compas installation problem

Voulp, 2021-01-04:
Hello,

I cannot understand what could I be doing so wrong, but even the simplest compas installation (using conda) is not finishing correctly. I created a new environment in conda ("research") and installed compas there, and I can see it in the Anaconda Navigator (compas 0.19.3), however when I load the environment in the Anaconda prompt and type "python -m compas" I get the following message:

C:\Users\kvoulpio\.conda\envs\research\python.exe: No module named compas.__main__; 'compas' is a package and cannot be directly executed

Moreover, when I go manually into C:\Users\<username>\.conda\envs\research I do not see any compas folders/files.

When I try to install the pip way, I get a command error with exit status 1 (see full text below)

Final note: when I load compas in a python terminal (in this case just the Windows cmd prompt) I get this:
>>> import compas
>>> compas.__version__
'0.5.0'

I can only think this is from an old installation (I briefly used compas two years ago). I cannot seem to find it anywhere in my system though (am I that stupid?). I have an Admin account on my machine, which has caused problems in the past (double installations etc). However in this case I installed everything on my user profile, so this should not be an issue.

Thank you in advance for advising on this issue. I use Windows 10, Python 3.9.1, and Anaconda 3 (which in turn has installed Python 3.8.6 on its environments.

***
cmd error from compas pip installation
***

    ERROR: Command errored out with exit status 1:
     command: 'c:\users\<username>\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\<username>\\AppData\\Local\\Temp\\pip-install-btuvgs6l\\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\\setup.py'"'"'; __file__='"'"'C:\\Users\\<username>\\AppData\\Local\\Temp\\pip-install-btuvgs6l\\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\<username>\AppData\Local\Temp\pip-record-4u7oudfb\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\<username>\appdata\local\programs\python\python39\Include\llvmlite'
         cwd: C:\Users\kvoulpio\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\
    Complete output (27 lines):
    running install
    running build
    got version from file C:\Users\kvoulpio\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\llvmlite/_version.py {'version': '0.34.0', 'full': 'c5889c9e98c6b19d5d85ebdd982d64a03931f8e2'}
    running build_ext
    c:\users\kvoulpio\appdata\local\programs\python\python39\python.exe C:\Users\kvoulpio\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py
    Trying generator 'Visual Studio 15 2017 Win64'
    Traceback (most recent call last):
      File "C:\Users\<username>\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py", line 191, in <module>
        main()
      File "C:\Users\<username>\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py", line 179, in main
        main_win32()
      File "C:\Users\<username>\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py", line 88, in main_win32
        generator = find_win32_generator()
      File "C:\Users\<username>\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py", line 76, in find_win32_generator
        try_cmake(cmake_dir, build_dir, generator)
      File "C:\Users\<username>\AppData\Local\Temp\pip-install-btuvgs6l\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\ffi\build.py", line 28, in try_cmake
        subprocess.check_call(['cmake', '-G', generator, cmake_dir])
      File "c:\users\<username>\appdata\local\programs\python\python39\lib\subprocess.py", line 368, in check_call
        retcode = call(*popenargs, **kwargs)
      File "c:\users\<username>\appdata\local\programs\python\python39\lib\subprocess.py", line 349, in call
        with Popen(*popenargs, **kwargs) as p:
      File "c:\users\<username>\appdata\local\programs\python\python39\lib\subprocess.py", line 947, in __init__
        self._execute_child(args, executable, preexec_fn, close_fds,
      File "c:\users\<username>\appdata\local\programs\python\python39\lib\subprocess.py", line 1416, in _execute_child
        hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
    FileNotFoundError: [WinError 2] The system cannot find the file specified
    error: command 'c:\\users\\<username>\\appdata\\local\\programs\\python\\python39\\python.exe' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\<username>\appdata\local\programs\python\python39\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\<username>\\AppData\\Local\\Temp\\pip-install-btuvgs6l\\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\\setup.py'"'"'; __file__='"'"'C:\\Users\\<username>\\AppData\\Local\\Temp\\pip-install-btuvgs6l\\llvmlite_c0c2d263e67f4b5498f97453d8c6bf5b\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\<username>\AppData\Local\Temp\pip-record-4u7oudfb\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\<username>\appdata\local\programs\python\python39\Include\llvmlite' Check the logs for full command output.

tetov, 2021-01-04:
Just to make sure, does the following work?
```
conda create --name test --yes
conda activate test
python
import compas
```

Voulp, 2021-01-04:
Hi,
Same error (No module named compas.__main__)
However I noticed that upon creating the new environment I load Python 3.9.1 (which already exists in my system). When I installed compas via conda, Python downgraded to 3.8.6. No idea if this could cause issues.

tetov, 2021-01-04:
Could you please post the output of these commands
```
conda activate test
where python
conda list
```
please? 

compas does not support Python 3.9 (yet) which is why anaconda is downgrading the python version in the environment.

Voulp, 2021-01-04:
(test) C:\>where python
C:\Users\kvoulpio\.conda\envs\test\python.exe
C:\Users\kvoulpio\AppData\Local\Programs\Python\Python39\python.exe
C:\Users\kvoulpio\AppData\Local\Microsoft\WindowsApps\python.exe

(test) C:\>conda list
# packages in environment at C:\Users\kvoulpio\.conda\envs\test:
#
# Name                    Version                   Build  Channel
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
ca-certificates           2020.12.5            h5b45459_0    conda-forge
certifi                   2020.12.5        py38haa244fe_0    conda-forge
compas                    0.19.3           py38haa244fe_0    conda-forge
contextlib2               0.5.5                      py_2    conda-forge
cycler                    0.10.0                     py_2    conda-forge
cython                    0.29.21          py38h4b9bc1a_1    conda-forge
decorator                 4.4.2                      py_0    conda-forge
ezdxf                     0.15             py38hbd9d945_2    conda-forge
freetype                  2.10.4               h546665d_0    conda-forge
imageio                   2.9.0                      py_0    conda-forge
importlib-metadata        3.3.0            py38haa244fe_2    conda-forge
importlib_metadata        3.3.0                hd8ed1ab_2    conda-forge
intel-openmp              2020.3             h57928b3_311    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsonschema                3.2.0                      py_2    conda-forge
kiwisolver                1.3.1            py38hbd9d945_0    conda-forge
laspy                     1.7.0              pyh5ca1d4c_0    conda-forge
lastools                  20171231          h0e60522_1001    conda-forge
laszip                    3.4.3                h6538335_1    conda-forge
libblas                   3.9.0                     6_mkl    conda-forge
libcblas                  3.9.0                     6_mkl    conda-forge
liblapack                 3.9.0                     6_mkl    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
libtiff                   4.2.0                hc10be44_0    conda-forge
llvmlite                  0.35.0           py38h57a6900_0    conda-forge
lz4-c                     1.9.3                h8ffe710_0    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6    conda-forge
m2w64-gcc-libs            5.3.0                         7    conda-forge
m2w64-gcc-libs-core       5.3.0                         7    conda-forge
m2w64-gmp                 6.1.0                         2    conda-forge
m2w64-libwinpthread-git   5.0.0.4634.697f757               2    conda-forge
matplotlib-base           3.3.3            py38h34ddff4_0    conda-forge
mkl                       2020.4             hb70f87d_311    conda-forge
mpmath                    1.1.0                      py_0    conda-forge
msys2-conda-epoch         20160418                      1    conda-forge
networkx                  2.5                        py_0    conda-forge
numba                     0.52.0           py38h4c96930_0    conda-forge
numpy                     1.19.2           py38hdf1ac2f_1    conda-forge
olefile                   0.46               pyh9f0ad1d_1    conda-forge
openssl                   1.1.1i               h8ffe710_0    conda-forge
pillow                    8.1.0            py38hf7ce48b_0    conda-forge
pip                       20.3.3             pyhd8ed1ab_0    conda-forge
planarity                 3.0.0.5           hfa6e2cd_1002    conda-forge
pycollada                 0.7.1                      py_0    conda-forge
pyparsing                 2.4.7              pyh9f0ad1d_0    conda-forge
pyrsistent                0.17.3           py38h294d835_1    conda-forge
python                    3.8.6           h7840368_4_cpython    conda-forge
python-dateutil           2.8.1                      py_0    conda-forge
python-planarity          0.4.1            py38h1e8a9f7_4    conda-forge
python_abi                3.8                      1_cp38    conda-forge
pyyaml                    5.3.1            py38hab1e662_1    conda-forge
schema                    0.7.3              pyh9f0ad1d_0    conda-forge
scipy                     1.6.0            py38h5f893b4_0    conda-forge
setuptools                49.6.0           py38h9bdc248_2    conda-forge
six                       1.15.0             pyh9f0ad1d_0    conda-forge
sqlite                    3.34.0               h8ffe710_0    conda-forge
sympy                     1.7.1            py38haa244fe_0    conda-forge
tk                        8.6.10               h8ffe710_1    conda-forge
tornado                   6.1              py38h294d835_0    conda-forge
vc                        14.2                 hb210afc_2    conda-forge
vs2015_runtime            14.28.29325          h5e1d092_0    conda-forge
watchdog                  1.0.2            py38haa244fe_0    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
wincertstore              0.2             py38h32f6830_1005    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
zipp                      3.4.0                      py_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.8                h4e2f164_1    conda-forge

tetov, 2021-01-04:
I have the exact same packages installed in my test env and I can still get `python -m compas` to work..

Might also be useful, @Voulp:
```
conda activate test
conda info
```

@brgcode @gonzalocasas could you please take a look?

gonzalocasas, 2021-01-04:
Could it be you have `compas 0.5.0` installed via an older method, like, adding it to the `PYTHONPATH` (environment variables)?

brgcode, 2021-01-04:
can you paste here the output of

```bash
echo %PATH%
```

Voulp, 2021-01-04:
Thank you both, I managed to find and delete all the old compas files from my 2019 manual install. Now it seems to work.

Just in case you want to see my %PATH%, here is the result (from the research env)

(research) C:\>python
Python 3.8.6 | packaged by conda-forge | (default, Dec 26 2020, 04:30:06) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import compas
>>> compas.__version__
'0.19.3'
>>> exit()

(research) C:\>echo %PATH%
C:\Users\<username>\.conda\envs\research;
C:\Users\<username>\.conda\envs\research\Library\mingw-w64\bin;
C:\Users\<username>\.conda\envs\research\Library\usr\bin;
C:\Users\<username>\.conda\envs\research\Library\bin;
C:\Users\<username>\.conda\envs\research\Scripts;
C:\Users\<username>\.conda\envs\research\bin;
C:\ProgramData\Anaconda3\condabin;
C:\SIMULIA\Commands;
C:\Program Files\Microsoft MPI\Bin;
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;
C:\ProgramData\Oracle\Java\javapath;
C:\WINDOWS\system32;
C:\WINDOWS;
C:\WINDOWS\System32\Wbem;
C:\WINDOWS\System32\WindowsPowerShell\v1.0;
C:\Program Files\MATLAB\R2019b\runtime\win64;
C:\Program Files\MATLAB\R2019b\bin;
C:\Program Files\MATLAB\R2018a\runtime\win64;
C:\Program Files\MATLAB\R2018a\bin;
C:\Program Files\Microsoft VS Code\bin;
C:\Program Files\dotnet;
C:\ProgramData\Anaconda3\Lib;
C:\WINDOWS\System32\OpenSSH;
C:\Program Files\Intel\WiFi\bin;
C:\Program Files\Common Files\Intel\WirelessCommon;
C:\Program Files (x86)\OpenSees3.1.0-x64.exe\bin;
C:\Program Files\PuTTY;
C:\WINDOWS\system32;
C:\WINDOWS;
C:\WINDOWS\System32\Wbem;
C:\WINDOWS\System32\WindowsPowerShell\v1.0;
C:\WINDOWS\System32\OpenSSH;
C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;
C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;
C:\Users\<username>\AppData\Local\Programs\Python\Python39\Scripts;
C:\Users\<username>\AppData\Local\Programs\Python\Python39;
C:\Users\<username>\AppData\Local\Microsoft\WindowsApps;
C:\Users\<username>\AppData\Local\Programs\MiKTeX 2.9\miktex\bin\x64;
C:\Users\<username>\AppData\Local\GitHubDesktop\bin;
C:\Users\<username>\AppData\Local\Microsoft\WindowsApps;
C:\Users\<username>\AppData\Local\Programs\Git\cmd

closed 2021-01-05 by brgcode