compas-dev/compas #730: RobotModel.data doesn't contain all necessary data

beverlylytle, 2021-01-06:
**Describe the bug**
Some data  is missing from in `RobotModel.data` making the serialization of instances of this class faulty.  In particular, if a `RobotModel` is loaded from a urdf that contains `gazebo` or other nonstandard tags, some information will be lost upon serialization followed by deserialization.  This is exposed by attempting to write the reserialized `RobotModel` to URDF.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the following script:
```
import copy

from compas.files import URDF
from compas.robots import RobotModel
import compas_fab

filepath = compas_fab.get('universal_robot/ur_description/urdf/ur5.urdf')
model = RobotModel.from_urdf_file(filepath)
model_copy = RobotModel.from_data(model.data)
# model_copy = copy.deepcopy(model)
urdf = URDF.from_robot(model_copy)
```
(Though this script imports `compas_fab`, it is only for accessing a particular URDF.  The relevant code lies in `compas`.)

**Expected behavior**
This should result in the creation of  a valid `URDF` instance, however an exception is thrown about a `URDFGenericElement` not having a tag.  This is because the protected attribute `_urdf_source` (or the inferred attribute `tag`)  is not included in `data`.

brgcode, 2021-11-09:
is this still a thing?

beverlylytle, 2021-11-09:
Yes.

closed 2024-04-23 by tomvanmele