compas-dev/compas #732: DefaultMeshLoader doesn't handle file uris correctly on Unix

beverlylytle, 2021-01-07:
**Describe the bug**
When loading meshes from local files on a Unix system, the `DefaultMeshLoader` takes the file uri `file:///path/to/mesh.obj` and strips it to `path/to/mesh.obj` making a relative path out of an absolute one, causing troubles.  While tinkering with the mesh loaders, it might also be good to look at whether it is still true that only obj files can be loaded remotely.

**To Reproduce**
Steps to reproduce the behavior:
Replacing the relative path as appropriate, run the following:
```
import compas
from compas.robots import DefaultMeshLoader

loader = DefaultMeshLoader()
assert loader.can_load_mesh('file://' + compas.get('boxes.obj'))
assert not loader.can_load_mesh('file:///relative/path/to/boxes.obj')
```

**Expected behavior**
I would expect both of these assertions to pass.  

**Desktop (please complete the following information):**
 - OS: OSX
 - Python version 3.8
 - Python package manager conda

beverlylytle, 2021-06-23:
[image]

tomvanmele, 2025-10-29:
closing as `compas.robots` no longer exists