compas-dev/compas #758: The version numbers don't match

ChiaChing-Yen, 2021-01-22:
The version number displayed in conda doesn't match the one comes from __version__.

### Steps to reproduce the behavior
1. Open Powershell, activate the env with COMPAS 1.0.0 installed
2. `conda list` and it displays compas 0.19.3
3. `python` `import compas` `compas.__version__`, and it displays 1.0.0

### Expected behavior
`conda list` should also displays compas 1.0.0

### Desktop:
 - OS: Windows 10 20H2
 - Python version: 3.7.9
 - Python package manager: conda

### Additional text
Congradulation for the release of the first stable version!

gonzalocasas, 2021-01-22:
This sounds like a local issue. Could you please try the following the same but use the anaconda prompt or the regular command prompt, instead of Powershell? At least when used in combination with VS Code, Powershell fails to activate environments properly, and this might be also happening to you.

If not sure, please post the full output of all your commands from activation to the `import compas` lines inside the python interpreter.

ChiaChing-Yen, 2021-01-22:
Sorry I didn't make it clear, I used Anaconda Powershell Prompt(miniconda3)

I found a folder called "COMPAS-0.19.3.dist-info" in my /envs/env_name/lib/site-packages, and when it co-exists with 1.0.0, conda will display the compas version as 0.19.3. After I deleted that folder everything is working like expected. Thank you!

gonzalocasas, 2021-01-22:
Good to hear that! 👍

closed 2021-01-22 by gonzalocasas