compas-dev/compas #767: VS code IntelliSense cannot find Mesh class

duchaoyu, 2021-02-08:
Hi, brgcode. @lottilotte and I are working on CSD2 and we both find that VScode Intellisense cannot find the Mesh class (in my win VScode and her mac VScode). When we execute the code, the code works. However, when we write the code, VScode doesn't show any hints. 

`from compas.datastructures import Network
from compas.datastructures import Mesh
print(Network)
print(Mesh)`

For example, the VS code can find the Network class and hint the class methods. 
[image]
[image]

However, this Mesh class cannot be found. VS Code only finds the mesh module, but not the class. So, no class method is shown.
[image]
[image]

This is not really a bug or a problem, just a bit annoying for teaching. We don't know whether it's a problem in our VS code, or in COMPAS...  

What I have tried:
1. changing the name of files under `compas.datastructures.mesh`, which doesn't help. 
2. specifying the path. Finally, VScode shows the hint, but I guess this is not the best solution. 
[image]

brgcode, 2021-02-08:
which language server are you using?

if `PyLance` then this is indeed a known problem, which will go away by switching back to, for example, `jedi`

duchaoyu, 2021-02-08:
[image]
Thank you for your reply, Tom. I'm sorry that I don't know language server. Is it in the VScode settings?
If it is, then my language server was `Microsoft`... I changed it to `Jedi` and restarted VScode. VScode somehow changed the language server back to `Microsoft`... but now I see hints in my code.... 
[image]
However, this method doesn't work on @lottilotte 's mac....

lottilotte, 2021-02-08:
Hi,

It says that I’m using Jedi. Or is this not the right place to check?

Many thanks!

brgcode, 2021-02-08:
these are the global settings. it can be that you have a local setting that makes VS Code use a different server. check in `.vscode`

yck011522, 2021-03-10:
i have one dirty hack that Tom is not going to like it:
ignore compas' two level import system. and import Mesh class from the actual document. 

```
# this
from compas.datastructures.mesh import Mesh
# or this
from compas.datastructures.mesh._mesh import Mesh
```

works for my VScode setup with Pylance

lottilotte, 2021-03-10:
Many thanks for the suggestion! Unfortunately for teaching this is rather unpractical...

brgcode, 2021-03-10:
For me the `jedi` language server works perfectly. To activate, try the following steps and if that doesn't work we can do a zoom session...

1. open settings (the little gear wheel at the bottom left of the window + select settings)
2. search for "language server" on the settings page
3. in the the search results select "Python" in the category options
4. select `jedi` as the language server
5. reload the window to activate (open the command palette and select `Developer: Reload window`)

[image]

lottilotte, 2021-03-11:
Hi, Yes no this doesn’t work unfortunately… Many thanks for your help. What time suits you best for zoom?

brgcode, 2021-03-11:
now?

brgcode, 2021-03-15:
@duchaoyu @lottilotte please check pylance branch. problem should be solved...

brgcode, 2021-03-18:
solved since `1.2.0`

closed 2021-03-18 by brgcode