compas-dev/compas #777: Polygon.centroid()

mattiskoh, 2021-02-25:
**Describe the bug**
`Polygon.centroid()` returns an incorrect value of a polygon with 4 points, it returns a centroid that is way off..  (e.g. z-coordinates not in between max and min values of polygon points..)

**To Reproduce**
Steps to reproduce the behavior:
1. Python script:
`from compas.geometry.primitives import Polygon, Point
Polygon(Point(0.000, -5.536, 1977.967), Point(0.000, 144.352, -0.000), Point(0.000, -148.968, -0.000), Point(0.000, 264.624, 1977.967))

print(Polygon.centroid)`

`>>>>Point(0.000, -471.050, -7031.803)`

**Expected behavior**
can't be correct.. should be something like Point(0, 134, 985)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version 3.8.6
 - Python package manager conda
 - Compas version 1.0.0

brgcode, 2021-02-25:
nice investigation :)
will have a look...

brgcode, 2021-02-25:
maybe briefly, just FYI, the recommended way to import compas objects is to always to this from "the second level", which is where the public API is defined. these imports should never change. below that level we sometimes reshuffle things if that is necessary. so in this case the recommended option is
```python
from compas.geometry import Polygon, Point
```

mattiskoh, 2021-02-25:
> maybe briefly, just FYI, the recommended way to import compas objects is to always to this from "the second level", which is where the public API is defined. these imports should never change. below [...]

Thanks for the heads-up. I used to do it "correctly", but the problem for me was that I could never get VSCode to properly auto-complete... I remember discussing this extensively with @yck011522 and @gonzalocasas a while back.

brgcode, 2021-02-25:
> Thanks for the heads-up. I used to do it "correctly", but the problem for me was that I could never get VSCode to properly auto-complete... I remember discussing this extensively with @yck011522 and [...]

yes, i know, especially with PyLance this is indeed a problem. we are working on that...

brgcode, 2021-02-25:
about the centroid...
the issue is related to how the centroid of the polygon object is calculated.
```python
from compas.geometry import Polygon, Point
from compas.geometry import centroid_points, centroid_polygon

polygon1 = Polygon([Point(0.000, -5.536, 1977.967), Point(0.000, 144.352, -0.000), Point(0.000, -148.968, -0.000), Point(0.000, 264.624, 1977.967)])
polygon2 = Polygon([Point(0.000, 144.352, -0.000), Point(0.000, 264.624, 1977.967), Point(0.000, -5.536, 1977.967), Point(0.000, -148.968, -0.000)])
```

1. using the method of the object

```python
print(polygon1.centroid)
print(polygon2.centroid)
```

2. using `centroid_points`

```python
print(centroid_points(polygon1))
print(centroid_points(polygon2))
```

3. using `centroid_polygon`

```python
print(centroid_polygon(polygon1))
print(centroid_polygon(polygon2))
```

the output of 1 and 3 is the same, but i guess you were expecting the outcome of 2 :)

brgcode, 2021-02-25:
the polygon object uses `centroid_polygon` in the background to calculate the centroid, which considers the actual polygon surface rather than the corner points to avoid skewed results when there are many points on the boundary that are potentially clustered together and change the location of the centroid. but this method only makes sense if the polygon is not self-intersecting.

unfortunately (apparently), we have not yet established full equivalence between the older functional approach and the newer oo system. this is something that should be resolved

mattiskoh, 2021-02-25:
Ok thanks for the clarification! I indeed expected the outcome of 2.  I also just looked into the api ref. of the `centroid_polygon` function and there is a big disclaimer for self-intersecting polygons :) so... feel free to close to the issue !

[image]

brgcode, 2021-02-25:
will leave it open as a reminder that the current implementation can lead to unexpected results :)

brgcode, 2021-02-25:
@gonzalocasas needs to be discussed in the next dev meeting...