compas-dev/compas #787: Maximum height limitation

archz2, 2021-03-16:
I am trying to build a shell structure with height of 80 mm but the result gives me 27.1mm. 

How can I ensure about the maximum height limitation with respect to the shape in the plan?

[image]

[80 height, but result 27.1.zip](https://github.com/compas-dev/compas/files/6147262/80.height.but.result.27.1.zip)

The zip file has the rv2 file and the 3dm file.

gonzalocasas, 2021-03-16:
@brgcode this issue probably should move to https://github.com/BlockResearchGroup/compas-RV2, right?

brgcode, 2021-03-16:
yes indeed...

gonzalocasas, 2021-03-16:
i thought `Transfer issue` would work, but it seems it only works across repos of the same org. 🤔

archz2, 2021-03-16:
Oh. Really sorry. Please delete this thread.

 I have re-posted my query here. 

https://github.com/BlockResearchGroup/compas-RV2/issues/256

gonzalocasas, 2021-03-16:
😉

closed 2021-03-16 by gonzalocasas