compas-dev/compas #792: Can't use compas in blender 2.92

tetov, 2021-03-19:
```
>>> import compas
```
Works fine.

```
>>> import compas_blender
Traceback (most recent call last):
  File "<blender_console>", line 1, in <module>
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas_blender\__init__.py", line 22, in <module>
    from .utilities import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas_blender\utilities\__init__.py", line 81, in <module>
    from .drawing import (
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas_blender\utilities\drawing.py", line 7, in <module>
    from compas.geometry import centroid_points
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas\geometry\__init__.py", line 635, in <module>
    from .transformations import (
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas\geometry\transformations\__init__.py", line 18, in <module>
    from .transformations_numpy import *  # noqa: F401 F403
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\compas\geometry\transformations\transformations_numpy.py", line 5, in <module>
    from numpy import asarray
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\numpy\__init__.py", line 143, in <module>
    from . import lib
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\numpy\lib\__init__.py", line 45, in <module>
    __all__ += type_check.__all__
NameError: name 'type_check' is not defined
```

Doesn't, because:
```
>>> import numpy
Traceback (most recent call last):
  File "<blender_console>", line 1, in <module>
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\numpy\__init__.py", line 143, in <module>
    from . import lib
  File "C:\Users\a\scoop\apps\blender\current\2.92\python\lib\site-packages\numpy\lib\__init__.py", line 45, in <module>
    __all__ += type_check.__all__
NameError: name 'type_check' is not defined
```

tetov, 2021-03-19:
I'll try to look into this a bit.

tetov, 2021-03-21:
In [`numpy/lib/__init__.py`](https://github.com/numpy/numpy/blob/v1.19.2/numpy/lib/__init__.py) there's references to modules that were imported from using star import.

```
from .foo import *

__all__ = [foo.__all__]
```

I found some mentions of this quirk: [1](https://www.mail-archive.com/numpy-discussion@scipy.org/msg27119.html) [2](http://numpy-discussion.10968.n7.nabble.com/How-to-run-unittests-at-build-time-td829.html)

Adding "non-star" imports fixes this. E.g:
```
from .foo import *
from . import foo

__all__ = [foo.__all__]
```

After I added imports for all modules I could do the hello-world program in [Getting Started - Blender](https://compas.dev/compas/latest/gettingstarted/blender.html)

Added to `numpy/lib/__init__.py`
```
from . import type_check
from . import index_tricks
from . import function_base
from . import nanfunctions
from . import shape_base
from . import stride_tricks
from . import twodim_base
from . import ufunclike
from . import histograms
from . import polynomial
from . import utils
from . import arraysetops
from . import npyio
from . import financial
from . import arraypad
```

[image]

I'm not sure how to fix this without editing numpy modules though..

gonzalocasas, 2021-03-22:
> In [`numpy/lib/__init__.py`](https://github.com/numpy/numpy/blob/v1.19.2/numpy/lib/__init__.py) there's references to modules that were imported from using star import.

maybe just stating the obvious, but these imports haven't changed recently on `numpy`'s side. And this also worked on Blender 2.91, so, I'm lost as to what could have changed that causes this.

tetov, 2021-03-22:
Yeah, they've been there for quite some time..

Just tried this with 2.91.0 and 2.91.2 and got the same error.

@brgcode Was this the problem you had with `>=2.90`? or was it something else?

brgcode, 2021-03-22:
for me only `2.83` ever worked.
and it still does btw.

the error for `>=2.90` was not related to `numpy` in my case.
i can check again in the afternoon...

brgcode, 2021-03-22:
but i agree with @gonzalocasas that this doesn't look like a problem with COMPAS...

brgcode, 2021-03-22:
have you tried importing `numpy` before switching to the COMPAS install? is there a difference between `numpy` versions before and after?

gonzalocasas, 2021-03-22:
just for the record, on a clean (as per our docs) install of `blender 2.91.2` with `compas 1.1.0`, the example posted works:

[image]

However, I did notice just now that the console lists heaps of errors during start up, but I could not capture them, they start with a complaint about `multiprocessing` potentially not working. I'll try to get to that log and post here.

gonzalocasas, 2021-03-22:
correction: not entirely clean, I also installed Sverchok on it.

Here's the system console log, the messages don't sound very alarming: 
```
Read prefs: C:\Users\gcasas\AppData\Roaming\Blender Foundation\Blender\2.91\config\userpref.blend
found bundled python: C:\Program Files\Blender Foundation\Blender 2.91\2.91\python
Unable to find the python binary, the multiprocessing module may not be functional!
sv: import settings
geomdl package is not available, some NURBS related nodes will not be available
SciKit-Image package is not available; SciKit-based implementation of Marching Cubes and Marching Squares will not be available
PyMCubes package is not available. PyMCubes-based implementation of Marching Cubes will not be available
Circlify package is not available. Circlify node will not be available
FreeCAD package is not available, Solids nodes will not be available
Dependencies available: pip, scipy, Cython.
C:\Users\gcasas\AppData\Roaming\Blender Foundation\Blender\2.91\scripts\addons\sverchok-master\settings.py:16: RuntimeWarning: Use 'sys.executable' instead of 'binary_path_python'!
  PYPATH = bpy.app.binary_path_python
sv: import all modules
C:\Users\gcasas\AppData\Roaming\Blender Foundation\Blender\2.91\scripts\addons\sverchok-master\settings.py:16: RuntimeWarning: Use 'sys.executable' instead of 'binary_path_python'!
  PYPATH = bpy.app.binary_path_python
sv: append app.handlers
sv: enable internal debug logging.
2021-03-22 11:21:04,332 [INFO] logging: log level, 20
 ______ _    _ _______  ______ _______ _     _  _____  _     _
/______  \  /  |______ |_____/ |       |_____| |     | |____/
______/   \/   |______ |    \_ |_____  |     | |_____| |    \_
initialized.

sv: version: 0.6.0.0
```

brgcode, 2021-03-22:
i am testing this on mac btw. could also make a difference...

tetov, 2021-03-22:
I did a clean install of 2.92 (using official installer,  removed Blender prefs from %APPDATA% and cleaned conda cache) and I get the same error. I'll try 2.91.2 next. Maybe there should be a note about "officially" supported blender version?

```
Traceback (most recent call last):
  File "<blender_console>", line 1, in <module>
  File "C:\Program Files\Blender Foundation\Blender 2.92\2.92\python\lib\site-packages\numpy\__init__.py", line 143, in <module>
    from . import lib
  File "C:\Program Files\Blender Foundation\Blender 2.92\2.92\python\lib\site-packages\numpy\lib\__init__.py", line 45, in <module>
    __all__ += type_check.__all__
NameError: name 'type_check' is not defined
```

brgcode, 2021-03-22:
> I did a clean install of 2.92 (using official installer, removed Blender prefs from %APPDATA% and cleaned conda cache) and I get the same error. I'll try 2.91.2 next. Maybe there should be a note about "officially" supported blender version?

i agree. think it should be `2.83` because that is the current LTS version...

tetov, 2021-03-24:
> 
> 
> just for the record, on a clean (as per our docs) install of `blender 2.91.2` with `compas 1.1.0`, the example posted works:
> 
>  _image removed for readability_
> 
> However, I did notice just now that the console lists heaps of errors during start up, but I could not capture them, they start with a complaint about `multiprocessing` potentially not working. I'll try to get to that log and post here.

I was able to reproduce this setup.
* `blender 2.91.2`
* `compas 1.1.0`
* `python 3.7.10`

but only after downgrading numpy: `1.19.2-py37he804e03_1 --> 1.17.5-py37hd20adf4_1` like you did for Sverchok. Otherwise I got same error as described in start of this issue.

gonzalocasas, 2021-03-24:
ok, so, basically if you pin the numpy version to the same one that blender ships with, things work?

tetov, 2021-03-24:
Seems like it. I'm in the process of testing this finding in my other installations.

brgcode, 2021-03-24:
would be cool if that consistently works.
however, somehow a bit annoying that one needs to make a separate install for blender because of the restrictions. 
or perhaps it is a good thing to sandbox that install in a separate environment anyway...

should we modify the install instructions to add a specific version of `numpy` to the `conda` environment?
or do we do this on the fly as part of the installation process?
with a warning...

gonzalocasas, 2021-03-24:
I think an environment that is built specifically for blender makes sense, we could prepare a couple of `environment.yml` files, one for each version of Blender that we support, it's not a lot of work and we'd get into a more consistent situation for sure.

tetov, 2021-03-24:
[image]

And here's a working setup for latest compas and latest Blender.

* Blender 2.92
* compas 1.2.1
* numpy 1.17.5
* python 3.7.10

brgcode, 2021-03-24:
> I think an environment that is built specifically for blender makes sense, we could prepare a couple of `environment.yml` files, one for each version of Blender that we support, it's not a lot of work and we'd get into a more consistent situation for sure.

i can make a section in the docs with "predefined installs" or something like that...

gonzalocasas, 2022-03-28:
@tetov although the currently documented process of install for Blender is problematic, we've found that doing a `pip install` using Blender's python seems to work properly most of the times for versions before Blender 3.1 (support for 3.1 will come in a while).

In a nutshell, the process that works for me always as of today is:
```bash
$ cd BLENDER_FOLDER/3.0/python/bin
$ python -m pip install compas
```

jf---, 2024-04-08:
@tetov since Blender has advanced so significantly, would you be willing to consider updating whether this still is an issue?
Blender is in such great shape that this issue still feels relevant.

tomvanmele, 2024-04-08:
the current instructions for using compas with Blender are here, and work very well afaik...

https://compas.dev/compas/latest/userguide/cad.blender.html

jf---, 2024-04-08:
@tomvanmele every time i see python + blender it just seems so shiny... 

but since the release discussed and the current version are so vastly different + the encouraging message above --  🔒 plz

closed 2024-04-09 by tetov