compas-dev/compas #794: API reference issues -

mattiskoh, 2021-03-23:
**Describe the bug**
More than once I have encountered that certain COMPAS functions are not shown in the API Reference, although they have already been implemented and can be found in the source code. This is slightly annoying :( . 

e.g. for `intersection_polyline_plane`
[image]
[image]
[image]
clearly it is implemented @ line 318 ->  compas/geometry/intersections/intersections.py

will add additional functions as I encounter them.. 

On the other hand, functions that have not been implemented can be found in the API search...  e.g. `trimesh_slice`

brgcode, 2021-03-23:
it is indeed possible that the API documentation is not always entirely up to date. will try to fix...

as for the slice function. it is a pluggable, meaning that the implementation is defined somewhere else (for example because it relies on functionality that is not part of python core). in this case, one possible implementation is in `compas_cgal`. so if you have `compas_cgal` installed the function should work...

perhaps we need a better way to document this :)

gonzalocasas, 2021-03-24:
would it be possible to add a unit test that cross-check the API docs against the API recorded in the json file (of the recently added unit test for api stability)? basically, if sphinx has an api to access the parsed TOC, we should at least ensure that all symbols are present in the TOC.

mattiskoh, 2021-04-16:
just also found that `intersection_segment_polyline` is also not showing up

gonzalocasas, 2022-03-28:
I think this is now much better, at least for the API stuff, search now includes all of them and we've been cleaning up and ensuring all functions are actually exposed, so, perhaps this can be closed, @mattiskoh ?

closed 2022-03-28 by mattiskoh