compas-dev/compas #808: network_find_cycles does not find faces in Mesh.from_lines()

lottilotte, 2021-04-08:
Hi,  I'm trying to create a mesh from a network. However, it doesn't work how I would like it to:

This is my network:
[image]
This is how I would like it to be meshed:
[image]
But this is what I get:
[image]

It seems like the network_find_cycles(network) is not functional?

```
import os

from compas.datastructures import Network
from compas.datastructures.mesh._mesh import Mesh

# ==============================================================================
# Paths
# ==============================================================================

HERE = os.path.dirname(__file__)
DATA = os.path.abspath(os.path.join(HERE, '..', 'data'))
FILE_I = os.path.join(DATA, 'ccf_3d.json')

# ==============================================================================
# Network to Mesh
# ==============================================================================

network = Network.from_json(FILE_I)
# lines = network.to_lines()
# mesh = Mesh.from_lines(lines, delete_boundary_face=True)

# ==============================================================================
# test the definition
# ==============================================================================

from compas.datastructures import network_find_cycles

vertices = network.to_points()
faces = network_find_cycles(network)
**# the faces are not found properly**: 
**# the output is: [[13, 7, 12, 19, 20, 3, 29, 9, 32, 11, 17, 30, 15, 8, 0, 21, 14, 1, 23, 25, 31, 16, 13], [0, 8, 17, 6, 27, 30, 17, 8, 15, 30, 27, 6, 24, 5, 4, 22, 18, 10, 26, 28, 2, 7, 13, 16, 12, 7, 2, 28, 12, 16, 31, 19, 26, 10, 20, 25, 23, 3, 18, 22, 29, 3, 23, 1, 29, 22, 4, 9, 29, 1, 14, 9, 4, 5, 32, 9, 14, 21, 32, 5, 24, 11, 32, 21, 0], [0, 11, 24, 6, 17, 11, 0], [18, 3, 20, 10, 18], [12, 28, 26, 19, 12], [20, 19, 31, 25, 20]]
# but it should be a regular quad mesh**

mesh = Mesh.from_vertices_and_faces(vertices, faces)

mesh.delete_face(0)
mesh.cull_vertices()

# ==============================================================================
# Visualisation
# ==============================================================================

from compas_rhino.artists import NetworkArtist
from compas_rhino.artists import MeshArtist

artist = NetworkArtist(network, layer="Network")
artist.clear_layer()
artist.draw()

artist = MeshArtist(mesh, layer="Mesh")
artist.clear_layer()
artist.draw()
```

[ccf_3d.json.zip](https://github.com/compas-dev/compas/files/6277019/ccf_3d.json.zip)

Many thanks!

brgcode, 2021-04-08:
making a mesh like this only works if the network is planar in projection. just transform the lines using a bestfit frame and transform back once the mesh is created...

lottilotte, 2021-04-08:
ok many thanks, this works nicely for one fold:

[image]

however, for multiple folds I will have to split it for the meshing I guess...
[image]

closed 2021-04-08 by lottilotte

brgcode, 2021-04-08:
how do you get to the folded line geometry to begin with?
if you can base the meshing on an unfolded version the solution is similar as above.
not that you only need to know the topology for the mesh generation.
the geometry can be snapped to any configuration afterwards...

brgcode, 2021-04-08:
also you can modify the face finding algorithm to only take the local neighbourhood of a node into account instead of the global geometry...

lottilotte, 2021-04-08:
> to only take the local neighbourhood of a node into account instead of the global geometry...

wouldn't this be ideal in general or is there a reason why one would want the global geometry?

brgcode, 2021-04-08:
more complicated and a bit slower, but yes...

lottilotte, 2021-04-08:
indeed seems complicated to me :see_no_evil: