compas-dev/compas #825: Feature Request: mesh_unweld(mesh, min_angle)

yck011522, 2021-04-30:
# Feature Request

As a [user], I want _[to have mesh_unweld function_] so that [_I can unweld meshes_].

## Details

At the moment we have `mesh_weld`, `meshes_join`, `meshes_join_and_weld`. Can we add also `mesh_unweld(mesh, min_angle)`
The `min_angle` is to check the angles for edges between two neighboring faces. If that angle is larger than the number spcified, it will be unwelded. Similar behavior can be found in Rhino / GH Unweld function.

brgcode, 2021-04-30:
it is there, but unfortunately undocumented
https://github.com/compas-dev/compas/blob/main/src/compas/datastructures/mesh/core/operations/weld.py

going through the entire API now...

yck011522, 2021-04-30:
oh cool. feel free to close if you see fit. or wait till doc is back.