compas-dev/compas #831: mesh_explode command

nizartaha, 2021-05-06:
**Describe the bug**
mesh_explode in compas seems that it doesnt behave as rhino explode command when exploding a closed mesh (boxy sort of typology). mesh is exported as stl from rhino (i tried also to export it as obj).

I am not sure if that is a bug or a feature request.

**To Reproduce**
Steps to reproduce the behavior:
```
import os
from compas.datastructures import Mesh, mesh_explode

HERE = os.path.dirname(__file__)
file_mesh = os.path.join(HERE, 'boxyMesh.stl')

mesh = Mesh.from_stl(file_mesh) 
mesh_parts = mesh_explode(mesh)
print(len(mesh_parts))
```

**Expected behavior**
in Rhino it returns 8 meshes and in compas it returns a list of only 1 element.

**Screenshots**
[image]

gonzalocasas, 2021-05-06:
@nizartaha could you please attach the STL you used as example?

brgcode, 2021-05-06:
can you send me the STL?

brgcode, 2021-05-06:
haha

nizartaha, 2021-05-06:
haha, thank you for the quick response. i tried to drop it actually but it didnt work.
here is it: https://we.tl/t-3ytvrFHSPk

brgcode, 2021-05-06:
Rhino explodes a mesh along components separated by naked edges (i.e. into disconnected components).

the problem is that the STL reader automatically merges identical vertices. this means that the mesh no longer has disconnected components en therefore cannot be exploded...

closed 2021-05-07 by nizartaha