compas-dev/compas #833: Non-admin install

gonzalocasas, 2021-05-11:
Following up on https://github.com/compas-dev/compas_fab/issues/297, it would be good if we can make sure that a non-admin user can still install COMPAS. There are two parts (at least), one is supporting Rhino to be installed in non-standard locations, and the other one is the need for admin rights to create symlinks.

jf---, 2024-04-08:
@gonzalocasas conda addresses this?

tetov, 2024-04-09:
This is not a problem for Py3 installs on Rhino 8. :)

tomvanmele, 2024-04-09:
has nothing to do with `conda`, but simply with how packages were/had to be installed in Rhino 7 and below. will close this since this already partially resolved and will only get better from now on...

closed 2024-04-09 by tomvanmele