compas-dev/compas #846: Feature Request: bestfit_line(points)

yck011522, 2021-06-02:
# Feature Request

As a [user], I want [to analyse some measured 3D points and fit them to a line] so that [benefit].

## Details

**Is your feature request related to a problem? Please describe.**
I was trying to find a bestfit line function in compas.
Turns out there are the more complex `bestfit_frame_numpy`, `bestfit_plane`, `bestfit_plane_numpy `and `bestfit_sphere_numpy`, but not bestfit_line.

**Describe the solution you'd like**
Just similar to other fitting analysis, I assume it can be a numpy implementation.

**Describe alternatives you've considered**
I put together something myself using numpy copying some code I found online:
```
data = np.array(measured_points)
datamean = data.mean(axis=0)
uu, dd, vv = np.linalg.svd(data - datamean)
line_vector = vv[0]
```

**Additional context**
Add any other context or screenshots about the feature request here.

beverlylytle, 2021-06-09:
Another option is `Polynomial.fit` with `deg=1` https://numpy.org/doc/stable/reference/generated/numpy.polynomial.polynomial.Polynomial.fit.html#numpy.polynomial.polynomial.Polynomial.fit

brgcode, 2021-06-09:
could you not just use `compas.numerical.pca_numpy` and use the origin and first axis of the pca?

brgcode, 2021-06-09:
i guess we should indeed provide a wrapper as with the other bestfit functions

brgcode, 2021-06-09:
i can send a PR later today so you can test if that is what you are looking for

yck011522, 2021-06-09:
Hi, two things to add to the feature request (now I start to use my own implementation more):

It would be even nicer if there could be both `bestfit_line` and `bestfit_line_segment`
The line segment version would return a segment that covers the min and max closest point projected to the line.

In general, it seems to make sense that these functions take unsorted points input. But it would be nicer if they will give a consistent line direction when faced with a sorted points input. This can save a lot of user code when using these functions to analyze measurement data.