compas-dev/compas #854: Expose draw_breps() at compas_rhino.utilities level and in documentation

yck011522, 2021-06-09:
I discovered this function that seems to work perfectly fine, although it is not used anywhere else in the core library. 
But I realize it is not in the `__all__` list. 

I want to ask if this function is intended to continue to exist in the future? And if so, can this be included in the `__all__` list

https://github.com/compas-dev/compas/blob/d2ba3a237a548b6bd396cc00913cae03a7189044/src/compas_rhino/utilities/drawing.py#L423-L511

yck011522, 2021-06-09:
I can _try_ create a PR for review.

brgcode, 2021-06-09:
equally (or even more) important than including it in those lists is to add it to the import statement in `compas_rhino.utilities.__init__.py` at line 176...

PR #857 (2021-07-05): Changes to draw_breps() to fix #854 #855

Fix #854 
Fix #855 

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [x] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're unsure about any of them, don't hesitate to ask. We're here to help! This is simply a reminder of what we are going to look for before merging you [... cut]
commit f70798db (2021-07-05): Merge pull request #857 from yck011522/main

Changes to draw_breps() to fix #854 #855
closed 2021-07-05 by brgcode