compas-dev/compas #855: Adding function to join the breps when calling draw_breps()

yck011522, 2021-06-09:
# Feature Request

As a [user who build my own RhinoArtist for my custom geometry classes], I want [to create polysurfaces/joinnedBreps using the draw_breps() function] so that [benefit].

## Details

**Is your feature request related to a problem? Please describe.**
Calling `compas_rhino.utilities.drawing.draw_breps()` with a list of input results in drawing multiple Breps. I want an option to create the the individual faces, join them, add them to doc and return the joined brep(s) guids.

**Describe the solution you'd like**
I think this can be backward compatible by using a flag.

brgcode, 2021-06-09:
can you add an example script of a use case scenario?

yck011522, 2021-06-09:
Yes, I can add an example to the docs to show that use case.

brgcode, 2021-06-09:
no i mean can you add an example here to clarify the context of the requested functionality?

yck011522, 2021-06-09:
Well, I added an example in the function's docs to show how to use:

    Using a compas Mesh as an example:
    >>> from compas.datastructures import Mesh
    >>> from compas.geometry import Box, Frame
    >>> from compas_rhino.utilities import draw_breps
    >>> box = Box(Frame.worldXY(), 1.0, 2.0, 3.0)
    >>> mesh = Mesh.from_shape(box)
    Draw convert each mesh face to brep dict schema:
    >>> vertices = [mesh.vertex_attributes(vertex, 'xyz') for vertex in mesh.vertices()]
    >>> breps = []
    >>> for face in mesh.faces():
    >>>     face = {'points' : [vertices[vertex] for vertex in mesh.face_vertices(face)]}
    >>>     face['points'].append(face['points'][0])
    >>>     breps.append(face)
    Draw brep faces as one joined brep.
    >>> guids = draw_breps(breps, join=True)

Context is a bit harder to say. Many times we need to model polysurfaces. One example is my timber beams with joint features. The features are 3d polygonal shapes. Not necessary extrusions, but can easily be modeled with polysurfaces composed of planar surfaces.

yck011522, 2021-06-09:
Arh, another example is for instance, if the existing compas_rhino.artist.PolyhedronArtist wants to change it's implementation from drawing the polyhedron as RhinoMesh objects to drawing it as Rhino Polysurface objects. 

Then you can simply use `draw_breps(breps, join=True)` as shown above to achieve it.

https://github.com/compas-dev/compas/blob/75072d9fcc1e2d0565c69d892ed16ead29b0f04a/src/compas_rhino/artists/polyhedronartist.py#L45-L82

PR #857 (2021-07-05): Changes to draw_breps() to fix #854 #855

Fix #854 
Fix #855 

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [x] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're unsure about any of them, don't hesitate to ask. We're here to help! This is simply a reminder of what we are going to look for before merging you [... cut]
closed 2021-07-05 by brgcode