compas-dev/compas #858: Mesh.from_polyhedron(polyhedron) should be renamed `from_platonicsolid`

yck011522, 2021-06-10:
## Expected Result

I expect `Mesh.from_polyhedron` to consume a Polyhedron class. Turns out it doesn't. This causes a slight semantic nuisance. 

## Actual Result

Currently, the function takes an integer and constructs a platonic solid and turns that to mesh.

I think the current thing should be renamed `Mesh.from_platonicsolid(f)` and the function `Mesh.from_polyhedron`  should follow the same implementation of `from_shape`. Note that I'm using Platonic solid

brgcode, 2021-06-10:
i guess that would make sense. perhaps i just provide an alias for now and a deprecation warning such that it is not a breaking change and can already be included now...

yck011522, 2021-06-10:
By the way, I got to say, the Polyhedron class is great for what I needed to do. For irregular solid models with planar faces.