compas-dev/compas #886: `compas_rhino.artists.RobotModelArtist.clear()` does not clear

yck011522, 2021-08-15:
**Describe the bug**
Can someone confirm if this is repeatable:

The `clear()` function in `compas_rhino.artists.RobotModelArtist` doesn't clear any geometry in the Rhino viewport.

https://compas.dev/compas/latest/api/generated/compas_rhino.artists.RobotModelArtist.clear.html#compas_rhino.artists.RobotModelArtist.clear

**To Reproduce**
Steps to reproduce the behavior:
1. Context [Rhino 6]
2. Load RobotModel from URDF
3. Something like this:
```
from compas.robots import RobotModel
from compas_rhino.artists import RobotModelArtist
# Drawin g is successful
robot_model = RobotModel.from_urdf_file(urdf)
robot_artist = RobotModelArtist(robot_model)
robot_artist.draw_collision()

# Try to clear the geometry is not successful

robot_artist.clear()
```

brgcode, 2021-11-09:
is this now resolved?

gonzalocasas, 2022-02-09:
WOMM

closed 2022-02-09 by gonzalocasas