compas-dev/compas #933: Versions in documentation

gonzalocasas, 2021-11-14:
The current docs have two issues regarding the versions dropdown:
 1. the sorting is alphabetical, so the newer versions like `1.11.1` are listed after `1.2.0` instead of above `1.9.3`
 2. the docs are defaulting to `1.7.1` instead of latest

Licini, 2021-11-29:
Hey @gonzalocasas 
1. I just added proper sorting in sphinx template (https://github.com/compas-dev/sphinx_compas_theme/commit/292090f07b5559d9c77843e2c44c06e96d72a116)
Should be working now if you clear the cache and load the doc site

2. @tomvanmele I think we decided `/latest` is not gonna be used anymore, instead `compas.dev/compas` will be directly forwarded to latest release and `dev` on top of drop list points to the latest commits from main branch. The current `latest` folder is actually leftover of old time which we should delete now.

3. I just released somehow it's not removing older patches anymore, and gh-pages branch might runout of space soon again.... I will make another PR on fixing this...
[image]

ChiaChing-Yen, 2021-11-30:
https://compas.dev/compas/latest/ is now giving a HTTP 404 error

brgcode, 2021-11-30:
yes we overlooked this :)

@licini

* `latest` is the latest released version. in this case `1.12.1`
* `dev` is the unreleased repo
* we only keep the latest patch of every released minor

last "rule" is not as important for now

brgcode, 2021-11-30:
we cannot solve this with a simple redirect to the latest version number because there are page links like https://compas.dev/compas/.../installation.index.html where this will not work

gonzalocasas, 2022-02-09:
this issue is solve afaics

closed 2022-02-09 by gonzalocasas