compas-dev/compas #936: RhinoMesh.from_geometry(M).to_compas() throws Runtime error (TypeErrorException)

ChiaChing-Yen, 2021-11-29:
**Describe the bug**
Collision Mesh Grasshopper component form COMPAS FAB doesn't work after recent updates. I tried isolating the issue and it turns out that `RhinoMesh.from_geometry(M).to_compas()` is throwing error. Same happens to `RhinoPoint.from_grometry(P).to_compas()`. Yet `RhinoMesh.from_guid(M).to_compas()` is working normally with the correct type hint.

**To Reproduce**
Steps to reproduce the behavior:
1. open Grasshopper Python script editor, set input name as 'M' and type hint as 'no type hint' or 'Mesh'
2. 
```python
from compas_rhino.geometry import RhinoMesh

a = RhinoMesh.from_geometry(M).to_compas()
```
3. input any Rhino mesh to the input 'M', I use a default mesh box from GH components.
4. 
```
  Error: Runtime error (TypeErrorException): unbound method to_compas() must be called with RhinoMesh instance as first argument (got nothing instead)

Traceback:
  line 3, in script
```

**Expected behavior**
It is supposed to return a compas mesh.

**Screenshots**
[image]

**Desktop:**
 - OS: macOS 12.0.1 and Win11 Enterprise
 - Python version: 3.7.12
 - Python package manager : conda 4.11

**Additional context**
 - Rhino 7.12.21313.06342
 - compas 1.12.0
 - compas_fab 0.20.1

brgcode, 2021-11-29:
this is indeed a bug and is fixed in the dev version on github since 6e5a48 but it is not released yet...

ChiaChing-Yen, 2021-11-29:
Good to know! Thank you COMPAS dev team for addressing this issue!

brgcode, 2021-11-29:
could you test version `1.12.1` and confirm that the problem is indeed resolved?
thanks

ChiaChing-Yen, 2021-11-30:
I updated to `1.12.1` and test the following methods with the steps I mentioned in the first comment:

* `RhinoPoint.from_geometry(M).to_compas()`
* `RhinoVector.from_geometry(M).to_compas()`
* `RhinoLine.from_geometry(M).to_compas()`
* `RhinoPlane.from_geometry(M).to_compas()`
* `RhinoMesh.from_geometry(M).to_compas()`

They all works as I expected, and the COMPAS FAB's components are working well, too.
I didn't test them with sophisticated test tools, yet from my point of view the problem are resolved.

brgcode, 2021-11-30:
cool thanks

closed 2021-11-30 by brgcode