compas-dev/compas #953: PluginNotInstalledError, NurbsCurve and NurbsSurface frome compas core

MacGros, 2021-12-15:
Hey,
User of orbingol's nurbs python, I'm more attracted to your future developpements with OCC thant ACIS :D Cheers !

But :

**Describe the bug**
Using the compas core geometry, a plugin is required to use NurbsCurve and NurbsSurface, the given url is inactive and the slug mention "factories", I've got no clue of what it might be :  https:/plugins.compas.dev/factories/new_nurbscurve_from_points

Obviously, a plugin is needed, but which one ? I've found [compas_nurbs ](https://gramaziokohler.github.io/compas_nurbs/latest/)in gk's github but that seems to be something entirely different.

**To Reproduce**
In vscode

```python
import numpy as np
from compas.geometry import Point, NurbsCurve, NurbsSurface

points = np.array([
    [0.0,  0.0, 0.0], 
    [10.0, 0.0, 0.0],
    [0.0, 10.0, 0.0], 
    [10.0,10.0, 0.0], 
    ])

nc2 = NurbsCurve.from_points( [Point(*points[2]), Point(*points[3])], degree=2)
```

```
Traceback (most recent call last):
  File "c:/_Archi/Etudes/21ASTRA_/4a_Segmentation-masonry/a1-piclouToTexture/a1-tests.py", line 23, in <module>
    nc1 = NurbsCurve.from_points( [p1,p2], degree=2)
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas\geometry\curves\nurbs.py", line 188, in from_points
    return new_nurbscurve_from_points(points, degree=degree)
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas\plugins.py", line 289, in wrapper
    plugin_impl = _select_plugin(extension_point_url)
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas\plugins.py", line 464, in select_plugin
    raise PluginNotInstalledError('Plugin not found for extension point URL: {}'.format(extension_point_url))
compas.plugins.PluginNotInstalledError: Plugin not found for extension point URL: https:/plugins.compas.dev/factories/new_nurbscurve_from_points
```

**Expected behavior**
Build a NurbsCurve from two points.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10
 - Python version : 3.7
 - Python package manager conda

**Additional context**

Packages in my conda env :
```
# packages in environment at C:\xxx\anaconda3\envs\geocc37:
#
# Name                    Version                   Build  Channel
argcomplete               1.12.3             pyhd8ed1ab_2    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
argon2-cffi               21.1.0           py37hcc03f2d_2    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bleach                    4.1.0              pyhd8ed1ab_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cffi                      1.15.0           py37hd8e9650_0    conda-forge
[... 157 lines cut ...]
traitlets                 5.1.1              pyhd8ed1ab_0    conda-forge
traittypes                0.2.1              pyh9f0ad1d_2    conda-forge
typing_extensions         4.0.1              pyha770c72_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
utfcpp                    3.2.1                h57928b3_0    conda-forge
vc                        14.2                 hb210afc_5    conda-forge
vs2015_runtime            14.29.30037          h902a5da_5    conda-forge
vtk                       9.0.1           no_osmesa_py37h96b1751_107    conda-forge
watchdog                  2.1.6            py37h03978a9_1    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.37.0             pyhd8ed1ab_1    conda-forge
widgetsnbextension        3.5.2            py37h03978a9_1    conda-forge
win32_setctime            1.0.4              pyhd8ed1ab_0    conda-forge
winpty                    0.4.3                         4    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_1    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.6.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h8ffe710_1013    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
```

Best Regards,
Antoine

romanarust, 2021-12-15:
Hi Antoine,

you need [compas_occ](compas_occ).
compas_nurbs will not be developed any further, as is replaced by the interface to occ.

brgcode, 2021-12-15:
hi, maybe a small thing to add to that is that the docs of `compas_occ` are a bit outdated. it is available on `conda-forge` now so you can just do

```bash
conda install -c conda-forge compas compas_occ --yes
```

MacGros, 2021-12-15:
Hey,

Thanks for the quick reply, installing compas_occ was the next step, I didn't understood that is was needed right now

After `conda install -c conda-forge compas compas_occ --yes` I had a TypeError on the Curve from lines example
```
Traceback (most recent call last):
  File "c:/_Archi/Etudes/21ASTRA_/4a_Segmentation-masonry/a1-piclouToTexture/a0-tests2.py", line 8, in <module>
    curve = NurbsCurve.from_line(line)
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas_occ\geometry\curves\nurbs.py", line 386, in from_line
    degree=1
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas_occ\geometry\curves\nurbs.py", line 216, in from_parameters
    curve = cls()
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas\geometry\curves\nurbs.py", line 96, in __new__
    return new_nurbscurve(*args, **kwargs)
  File "C:\Users\antgr\anaconda3\envs\geocc37\lib\site-packages\compas\plugins.py", line 292, in wrapper
    return plugin_impl.method(*args, **kwargs)
TypeError: new_nurbscurve() missing 1 required positional argument: 'cls'
```

So I installed the last pack : `pip install git+https://github.com/compas-dev/compas_occ.git`

And now, everything is fine
Cheers !

Antoine

ps : I've just seen your forum, https://forum.compas-framework.org/ which will be more suited for the next times :)