compas-dev/compas #954: RhinoWIP 8 on MAC : Runtime error (IOException): Could not add reference to assembly System.Windows.Forms.DataVisualization

weichengyuan1226, 2021-12-15:
Since the Rhino8 on Mac is moving from Mono (Microsoft’s .NET Framework.) to Microsoft.Net 6. the System.Windows.Forms.DataVisualization.Charting Namespace is not available anymore.  So in Mac RhinoWIP 8  Python editor run 'compas_rhino' it will return the error 'Could not add reference to assembly System.Windows.Forms.DataVisualization '

Steps to reproduce the behavior:

1. Mac version RhinoWIP 8 
2. `import compas_rhino`
3.  ```
Error: Runtime error (IOException): Could not add reference to assembly System.Windows.Forms.DataVisualization

Traceback:
  line 17, in <module>, "/Users/chengyuanwei/Library/Application Support/McNeel/Rhinoceros/8.0/scripts/compas_rhino/forms/chart.py"
  line 39, in <module>, "/Users/chengyuanwei/Library/Application Support/McNeel/Rhinoceros/8.0/scripts/compas_rhino/forms/__init__.py"
  line 14, in <module>, "/Users/chengyuanwei/Library/Application Support/McNeel/Rhinoceros/8.0/scripts/compas_rhino/utilities/misc.py"
  line 157, in <module>, "/Users/chengyuanwei/Library/Application Support/McNeel/Rhinoceros/8.0/scripts/compas_rhino/utilities/__init__.py"
  line 30, in <module>, "/Users/chengyuanwei/Library/Application Support/McNeel/Rhinoceros/8.0/scripts/compas_rhino/__init__.py"
  line 1, in script
```

<img width="752" alt="Screen Shot 2021-12-15 at 2 21 22 PM" src="https://user-images.githubusercontent.com/72129688/146251998-f2a55684-6ed6-4475-a013-b67ddcc14c06.png">
![Screen Shot 2021-12-15 at 2 19 21 PM copy](https://user-images.githubusercontent.com/72129688/146252464-036df092-08c2-41a0-ae0b-f4faf2d05f97.png)

 - OS: MacOS Monterey
 - Python version: 3.8.12
 - Python package manager: conda

weichengyuan1226, 2022-03-04:
Hi,

Did the latest COMPAS fix this issue?

brgcode, 2022-03-06:
not but i will take care of it now...

weichengyuan1226, 2022-04-02:
the issue is gone with 1.15.1. Thank you!

closed 2022-04-02 by brgcode