compas-dev/compas #977: Enable "conda install COMPAS" to work on Apple Silicon M1 Macs

irruputuncu, 2022-01-22:
# Feature Request

As an M1 mac user, it would be fantastic to be able to follow the suggested installation for compas without errors. Maybe it's as easy as adding a respective image to the `conda-forge/osx-arm64` channel.

## Details

When trying to install compas with the regular (mini)conda setup I get the error message that the compas package is not available in any of the channels, even after running `conda config --add channels conda-forge` (as expected since it is not in the ARM channel). I then tried to install the package via pip, which somewhat worked but only installed version `0.3.9`. 

The best workaround I could find was to set up conda in rosetta/intel-virtualization mode following these [instructions](https://github.com/Haydnspass/miniforge#rosetta-on-mac-with-apple-silicon-hardware) on a miniforge fork:

```
CONDA_SUBDIR=osx-64 conda create -n [environment]   # create a new environment
conda activate [environment]
conda env config vars set CONDA_SUBDIR=osx-64  # subsequent commands use intel packages
```

In that environment, the installation went as expected and I got the latest version. However, this of course runs all packages using the Intel versions which might impair performance.

brgcode, 2022-01-22:
could you provide some more details about your system?
i have M1 (max) and had absolutely no issue with installation using `miniconda`...

irruputuncu, 2022-01-22:
Strange, then we should have the same system. I use `miniconda` as well and installed it using the `Miniconda3 macOS Apple M1 64-bit bash (Py38 conda 4.10.1 2021-11-08)` link on their [website](https://docs.conda.io/en/latest/miniconda.html). Maybe that's already where the difference lies. If that is the case, I would guess that the conda/python using the "normal" miniconda installation is automatically running in Rosetta/Intel mode and using the respective `conda-forge/osx-64` conda channel to find the package. 
Otherwise my system is pretty fresh and I haven't fiddled with any other python environments on it before. So I am not sure what else could be the issue.

Interestingly enough, just adding the default osx channel `https://conda.anaconda.org/conda-forge/osx-64` to my default ARM conda environment does not seem to be enough for it to find the `COMPAS` package.

brgcode, 2022-01-22:
as far as i can tell, we are supposed to send a PR to the `conda-forge-pinning` feedstock to add support for `osx_arm64` to `compas`.

https://conda-forge.org/blog/posts/2020-10-29-macos-arm64/

i just did that.
see here https://github.com/conda-forge/conda-forge-pinning-feedstock/pull/2434.
i also tried the manual version but it failed.

i guess now we have to wait :)

brgcode, 2022-02-03:
@irruputuncu seems like this should be solved now :)

could you have a look and confirm?

thanks!

irruputuncu, 2022-02-03:
Yes! It is seamlessly installed from the ARM channels now and I can also confirm that python runs in Apple Silicon mode now.

Thanks a lot!

closed 2022-02-03 by irruputuncu