compas-dev/compas #993: Does not correctly detect IronPython in Rhino 7.16

cwensley, 2022-02-15:
In Rhino 7.16, we will be updating to use IronPython 2.7.12, which no longer reports "IronPython" in its `sys.version` which breaks [this code in _os.py](https://github.com/compas-dev/compas/blob/main/src/compas/_os.py#L99):

```py
def is_ironpython():
    """Check if the Python implementation is IronPython.
    Returns
    -------
    bool
        True if the implementation is IronPython. False otherwise
    """
    return 'ironpython' in sys.version.lower()
```

Instead, I would recommend this to be changed to check `platform.python_implementation()` instead (or in addition to) so it continues to work with future versions of IronPython and Rhino.

Please let me know if you need any more information about this problem.  For reference, the user running into this reported the issue to us [here](https://discourse.mcneel.com/t/rhino7-crash-on-loading-docstrings-of-build-in-funtion-in-ghpython/137348/4).

gonzalocasas, 2022-02-15:
@cwensley thanks for the report! Will fix asap!

gonzalocasas, 2022-02-16:
Fixed! To be released real soon

closed 2022-02-16 by gonzalocasas