compas-dev/compas #995: Blender installation does not work on Windows dus to inability to locate DLLs

brgcode, 2022-02-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Context [e.g. ST3, Rhino, Blender, ...]
2. Sample script
3. Sample data
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version [e.g. 2.7]
 - Python package manager [e.g. macports, pip, conda]

**Additional context**
Add any other context about the problem here.

brgcode, 2022-03-11:
the following worked on my windows laptop...
using the normal command line (not anaconda prompt)
with admin privileges

1. (this is just for convenience) add Blender `python\bin` to the `PATH`
2. make sure there are no other Pythons on the `PATH` (`where python`)
3. `python -m ensurepip`
4. `python -m pip install --upgrade pip`
5. `python -m pip install compas`

brgcode, 2022-03-11:
@Licini could you try this on your machine?

brgcode, 2022-03-11:
this might also be related
https://github.com/ContinuumIO/anaconda-issues/issues/11994

gonzalocasas, 2022-03-15:
Is it really needed to include `ensurepip` and `--upgrade pip`? Since Blender ships with a decently updated python, isn't it irrelevant and perhaps even negative to force an update?

brgcode, 2022-03-28:
no, but running this now again on a fresh windows installation, i had to even modify the prefix to make sure windows doesn't keep adding stuff to the roaming profile.

```bash
cd "C:\Program Files\Blender Foundation\Blender 2.93\2.93\python"
.\bin\python.exe -m pip install compas --prefix="." --force-reinstall
```

jf---, 2024-04-08:
> Since Blender ships with a decently updated python, isn't it irrelevant and perhaps even negative to force an update?

+1

@gonzalocasas lets keep weeding the tracker ;)

tomvanmele, 2024-04-08:
see https://compas.dev/compas/latest/userguide/cad.blender.html

closed 2024-04-08 by tomvanmele