compas-dev/compas #997: Suggestion: switch to black code formatter

gonzalocasas, 2022-03-05:
Considering the prominence of `black` (https://github.com/psf/black) as the increasingly de-facto standard for code formatting on Python, it might be a good idea to switch to it and give up all hand-crafted control of code style in favor of total consistency.

Reasons to use black:
- Our code is mostly `black` compatible already. I tried locally, and by far, the largest change is the consistent update of all string quotes to be double quoted.
- Consistency, consistency, consistency. More and more big (and small) projects use it, and the lack of configurability means all projects are styled consistency even across organizations.
- When working with code, you don't even have think about formatting, it will be done entirely for you. If you're an experienced coder, this is good because you can focus on what matters, if you are a novice coder, it's also good because it removes a whole category of review comments that need to be addressed later on.
- It's not configurable: you can have any color you want, as long as it's black.
- It's called black!

Reasons not to use black:
- [OCD](https://en.wikipedia.org/wiki/Obsessive%E2%80%93compulsive_disorder)  ;)

brgcode, 2022-03-06:
sure but perhaps this can be part of COMPAS 2.0?

gonzalocasas, 2022-03-06:
👍

jf---, 2024-04-08:
> [OCD](https://en.wikipedia.org/wiki/Obsessive%E2%80%93compulsive_disorder) ;)
LOL

@gonzalocasas this in integrated in the CI.
see #1208 for the suggesion of invoke `black` as per pre-commit

gonzalocasas, 2024-04-23:
`black` is already integrated, and we have https://github.com/compas-dev/compas/issues/1329 for the pre-commit hook

closed 2024-04-23 by gonzalocasas