compas-dev/compas #1002: Issue visualizing mesh with colors in compas ghpython

joburger, 2022-03-11:
I can't manage to correctly visualize a mesh with colors using compas ghpython in Grasshopper. Using Rhino, the artists work correctly but in ghpython different colors passed to the `draw_faces` or `draw_vertices` do not seem to have an effect. Am I missing something here?

**Expected behavior**
The mesh, mesh faces, and mesh vertices drawn in different colors.

**To Reproduce**
Rhino 7.15
compas 1.14.1

```
import compas_rhino
from compas_ghpython.artists import MeshArtist

rhinomesh = compas_rhino.conversions.RhinoMesh.from_geometry(inputMesh)
compasmesh = rhinomesh.to_compas()

artist = MeshArtist(compasmesh)
a = artist.draw_faces(color=[0, 0, 255], join_faces=True)
b = artist.draw_vertices(color=[0, 255, 0])
c = artist.draw(color=[255, 0, 0])
```

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows
 - Conda

brgcode, 2022-03-11:
no, i noticed there is indeed no (or very limited) support for custom colors in `compas_ghpython`. i managed to get something to work, but will take a bit of time to send out a proper PR...
[image]

joburger, 2022-03-11:
Alright thanks! Will try to find a workaround for now