compas-dev/compas #1037: gh_python.MeshArtist doesn't work

baolianliu, 2022-04-21:
**Describe the bug**
When I created a mesh in COMPAS_singular, I want to visualize the mesh by ghpython.MeshArtist.draw( ), it doesn't work. Is there anything missing?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows10, Rhino7
 - Python version 3.9.6
 - Python package manager [pip, conda]

gonzalocasas, 2022-04-21:
is `densemesh` a `compas.datastructures.Mesh`?

gonzalocasas, 2022-04-21:
also, please add the version of compas that you're using.

baolianliu, 2022-04-21:
> is `densemesh` a `compas.datastructures.Mesh`?

COMPAS version is 1.15.1

gonzalocasas, 2022-04-22:
In that case, i think this should be reported in the https://github.com/BlockResearchGroup/compas_singular repository

closed 2022-04-25 by baolianliu