compas-dev/compas #1092: Incorrect detection of Rhino

AdamAnouar, 2022-10-27:
**Describe the bug**
If [Rhino-stubs](https://pypi.org/project/Rhino-stubs/) are installed on the cpython environment, compas will incorrectly assume that it is inside Rhino (`compas.RHINO == True`)

To reproduce:
 * Install Rhino-stubs in an environment
 * Start python and run `from compas.geometry import Box`

chenkasirer, 2022-10-27:
`Rhino-stubs` wouldn't let me import `Rhino`.
After installing `rhinoinside` I can import `Rhino` but only after calling `rhinoinside.load()`. In that context, COMPAS is not entirely wrong as there is really an instance of Rhino available. What's missing are indeed `rhinoscriptsyntax` and `scriptcontext` which might cause certain things to break later on. 

But maybe we could test more accurately with `Rhino.__file__`:
```python
# in Rhino/GH
>import Rhino
>Rhino.__file__
RhinoCommon, Version=7.22.22255.5001, Culture=neutral, PublicKeyToken=552281e97c755530, Rhino.UI, Version=7.22.22255.5001, Culture=neutral, PublicKeyToken=552281e97c755530
```

```python
# in CPython with rhinoinside
>import Rhino
>Rhino.__file__
'C:\\Program Files\\Rhino 7\\System\\RhinoCommon.dll'
```

gonzalocasas, 2022-10-27:
Ok, this needs to be investigated further. When i tested on @AdamAnouar computer, `Rhino.__file__` was empty but `Rhino.__path__` was pointing to the site packages.