compas-dev/compas #1096: Error using GH Artist when Ladybug is installed

chenkasirer, 2022-11-09:
There appears to be some conflict related to COMPAS `Artist` and the Ladybug plugin.

Steps to reproduce the behavior:
1. Install COMPAS
2. Install Ladybug Tool from [here](https://www.food4rhino.com/en/app/ladybug-tools).
3. Open a `ghpython` component in Grasshopper.
4. Try to draw something with an Artist

For example:
```python3
from compas.geometry import Box
from compas.artists import Artist

box = Box.from_width_height_depth(5,5,5)
a = Artist(box).draw()
```

Which produces the following errors:

```
C:\Users\ckasirer\ladybug_tools\python\Lib\site-packages\pkg_resources\py2_warn.py:1: UserWarning: Setuptools no longer works on Python 2
************************************************************
Encountered a version of Setuptools that no longer supports
this version of Python. Please head to
https://bit.ly/setuptools-py2-sunset for support.
************************************************************
  import sys
```

```
Runtime error (SystemExitException): SystemExit

Traceback:
  line 86, in <module>, "C:\Users\ckasirer\ladybug_tools\python\Lib\site-packages\pkg_resources\__init__.py"
  line 13, in <module>, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_ui\rhino\install.py"
  line 13, in <module>, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fofin\rhino\install.py"
  line 423, in try_import, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\plugins.py"
  line 191, in load_plugins, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\plugins.py"
  line 155, in registry, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\plugins.py"
  line 492, in collect_plugins, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\plugins.py"
  line 311, in wrapper, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\plugins.py"
  line 16, in <module>, "C:\Users\ckasirer\ladybug_tools\python\Lib\site-packages\pkg_resources\py2_warn.py"
  line 97, in __new__, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\artists\artist.py"
  line 6, in script
```

**Expected behavior**
Drawing of the box.

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
The Ladybug plugin seems to download CPython3.7 and add its `site-packages` directory to its search path as part of its installation.

gonzalocasas, 2022-11-09:
The problem is the plugin system, it's trying to enumerate stuff over `site-packages` and basically triggering hell.

chenkasirer, 2022-11-09:
Do you think there's anything we could do about this or it's fully up to GH plugin behavior?
Also why is Ladybug downloading its own instance of python? did you get what's it using it for?

gonzalocasas, 2023-10-10:
Fixed on https://github.com/compas-dev/compas/pull/1185

closed 2023-10-10 by gonzalocasas