compas-dev/compas #1097: Strange result converting to/from rhino NURBS circle

chenkasirer, 2022-11-09:
When converting a circle shape NURBS curve from Rhino to COMPAS then back to Rhino the resulting shape seems distorted.

Steps to reproduce the behavior:
1. Create a Rhino.Geometry.Circle
2. Convert Circle to NURBS curve using `Circle.ToNurbsCurve()`
3. Convert NURBS curve to COMPAS RhinoNurbsCurve
4. Serialize and de-serialize
5. Draw the resulting NURBS Curve

For example, in Grasshopper:
```python3
from compas_rhino.geometry import RhinoNurbsCurve
import scriptcontext as sc

circle = sc.doc.Objects.Find(circle).Geometry
nurbs = circle.ToNurbsCurve()

compas_curve = RhinoNurbsCurve.from_rhino(nurbs)
curve_data = compas_curve.data
a = RhinoNurbsCurve.from_data(curve_data).rhino_curve
```
The result is a geometry different than the original circle.

**Expected behavior**
Should result in the same geometry

**Screenshots**
[image]

[image]

gonzalocasas, 2022-11-10:
Is this closed then?

chenkasirer, 2022-11-10:
Yup! fixed in https://github.com/compas-dev/compas/pull/1099.