compas-dev/compas #1107: Bug in plane conversions

ZacZhangzhuo, 2022-12-12:
**Describe the bug**
When using `plane_to_rhino` and `plane_to_compas`, the conversion is not correct. My guess is that Rhino only records Origin and Normal for describing a plane, which is not comprehensive.

Here is an example to of the bug:

`import Rhino.Geometry as rg
from compas_rhino.conversions import plane_to_compas
from compas_rhino.conversions import  plane_to_rhino

plane = rg.Plane(rg.Point3d(-99.0,-187.5,117.9),rg.Vector3d(-0.7,0.6,0),rg.Vector3d(-0.1,-0.1,0.9))

plane = plane_to_rhino(plane_to_compas(plane)) # comment or uncomment this line, the plane is different in Rhino screen`

[image]
[image]

chenkasirer, 2022-12-12:
Hi @ZacZhangzhuo, thanks for opening this issue!

Yes. It's actually `compas.geometry.Plane` that's only defined using a point and a z-normal vector, so when converting a `Rhino.Geometry.Plane` this information is indeed lost. Instead you could use `compas.geometry.Frame` which is defined using a point and two vectors, see in [docs](https://compas.dev/compas/latest/api/generated/compas.geometry.Frame.html#compas.geometry.Frame).

You can then convert to and from `Rhino.Geometry.Plane` using `plane_to_compas_frame` and `frame_to_rhino_plane` correspondingly. So slightly adapting your code, this would look like this:

```
import Rhino.Geometry as rg
from compas_rhino.conversions import plane_to_compas_frame
from compas_rhino.conversions import frame_to_rhino_plane

plane = rg.Plane(rg.Point3d(-99.0,-187.5,117.9),rg.Vector3d(-0.7,0.6,0),rg.Vector3d(-0.1,-0.1,0.9))
plane = frame_to_rhino_plane(plane_to_compas_frame(plane)) # comment or uncomment this line, the plane is different in Rhino screen
```

ZacZhangzhuo, 2022-12-12:
Thank you so much!

closed 2022-12-12 by ZacZhangzhuo