compas-dev/compas #1115: CircleArtist().draw() freezes views in MacOS Rhino

zenbagailu, 2023-02-02:
**Describe the bug**
When running the draw()  method of CircleArtist from the Python Editor in MacOS, the views becoming unresponsive (and it is then necessary to restart Rhino). The rest of the GUI, and even selection on the views. seems to work.

**To Reproduce**
Steps to reproduce the behavior:
Context: Rhino 7 and 8 (WIP) in MacOS (M1), Compas version 1.17.4.
Open the Rhino Python Editor (Tools->PythonScript->Edit...)
Copy and paste the following script (the code executes correctly if run directly through Tools->PythonScript->Run..)
``` 
from compas.geometry import Circle
from compas_rhino.artists import CircleArtist

circle = Circle([[0,0,0], [0,1,1]],20)
artist = CircleArtist(circle)

artist.draw()
```
Run it.
**Expected behavior**
The views in Rhino should  respond as normally after running the script.

**Desktop (please complete the following information):**
 - OS: macOS 13.1, Ventura
 - Python version: IronPython, 2.7
 - Python package manager: conda

**Additional context**
I observed the same behavior using Artist() instead of CircleArtist(), and  with MeshArtist and its draw(), draw_edges(), and draw_vertices() (with a Mesh object).

brgcode, 2023-02-02:
the built-in code editor on mac does not issue a redraw at the end of executing a script automatically. if you press escape, the view will unfreeze. or if you trigger a redraw explicitly with the artist at the end of the script.

btw, there is no need to import a specific artist for drawing compas objects. you should be able to do something like this.

```python
from compas.geometry import Circle
from compas.artists import Artist

circle = Circle([[0,0,0], [0,1,1]],20)
artist = Artist(circle)
artist.draw()

# trigger an explicit redraw
Artist.redraw()

# or press ESC after executing the script
```

brgcode, 2023-02-02:
note that this redraw thing is only needed when using the built-in code editor on mac. in all other scenarios it happens automatically...

zenbagailu, 2023-02-02:
That worked, thank you for the quick feedback!

closed 2023-02-02 by zenbagailu