compas-dev/compas #1119: Add plane_brep intersection method

ZacZhangzhuo, 2023-02-24:
# Feature Request

Add `plane_brep intersection` method

## Details
Just found in `compas.geometry`, there is no `intersection_plane_brep` like `Rhino.Geometry.Intersect.Intersection.BrepPlane` method in Rhino API. This function would be useful for slicing objects in additive printing, or other fabrication scenarios. Hope this function could be added.

## This request is also added in [compas forum](https://forum.compas-framework.org/t/add-plane-brep-intersection-method/672)

ZacZhangzhuo, 2023-02-24:
just found the `compas.geometry.Brep.slice` in [Brep ](https://compas.dev/compas/latest/api/generated/compas.geometry.Brep.slice.html#compas.geometry.Brep.slice)is exactly what we need, although it is a bit hidden and not categorized in `compas.geometry.intersection`. Hope the whole Brep class can be upgrade for easier use soon!

gonzalocasas, 2023-02-28:
/cc @chenkasirer

chenkasirer, 2023-02-28:
Commented on the discussion [here](https://forum.compas-framework.org/t/add-plane-brep-intersection-method/672) 
TL;DR there's a `trim()`, is it the same? is it just a naming difference?