compas-dev/compas #1126: Rhino crashing when trying to use compas for first time

mjnewsum, 2023-03-10:
**Describe the bug**
If I run a python script with `import compas`, rhino will freeze and crash.
If I type `from compas.` in the python editor, rhino will freeze and crash.
If I drop a compas component on the grasshopper canvas, rhino will freeze and crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Context - Rhino 7, Windows 10, Python 3.11.0, miniconda conda 23.1.0
2. import conda and run
3. Sample data - I don't get any error or crash data...
4. See error - Not sure where to see error.

**Expected behavior**
I would expect rhino python to be able to find the compas library, but it seems to fail to read the library. Not sure if I need to activate my conda environment from within rhino somehow.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python version: 3.11.0
 - Python package manager: miniconda conda 23.1.0

**Additional context**
This is my first time running rhino with compas after installing using the Installation and Getting Started - Rhino steps.

mjnewsum, 2023-03-10:
I just tried uninstalling compas using `python -m compas_rhino.uninstall` and I got the following error:

> (research) C:\Users\mnewsum>python -m compas_rhino.uninstall
Uninstalling COMPAS packages from Rhino 7.0 scripts folder:
C:\Users\mnewsum\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts

The following packages have been detected and will be uninstalled:

   compas               OK
   compas_ghpython      OK
   compas_rhino         OK
   compas_bootstrapper  OK

Running post-uninstallation steps...

One or more errors occurred:

   - AttributeError("'NoneType' object has no attribute 'version'")

Uninstall completed.

When I open rhino and launch grasshopper, I still see the compas tab. I did not expect to still have the userObjects. When I put a component on the canvas, it has the following error:

> No module named compas_bootstraps

mjnewsum, 2023-03-10:
Are there any other files I need to remove to completely uninstall compas manually?

I have checked the Scripts folder and it has all symlinks removed.
I have removed all userObjects from grasshopper.
I am not sure if there are any other files I might need to manually reset so install and uninstall work correctly.

mjnewsum, 2023-03-10:
I am not sure if this is related, but I noticed that `python -m compas` shows that compas is installed correctly, but my Python is `Python: 3.11.0 (CPython)`. on the installation page it shows Python as `Python: 3.8.2 | package by conda-forge`.

[image]

I have conda-forge included when I create my conda environment. `conda create -n robot -c conda-forge compas`

mjnewsum, 2023-03-14:
I have tried to install compas multiple times now. The installation process seems smooth and everything appears to be successful, but rhino crashes anytime I `import compas` or use one of the grasshopper components.

I am not sure what else I should try. I have uninstalled compas, uninstalled miniconda, uninstalled rhino, and started completely over. I tried installing anaconda instead of miniconda, and I still ended up with a build crashes rhino.

Is there anything I can check to confirm that the install is setup correctly?

mjnewsum, 2023-03-14:
I followed the Installation and Getting Started steps listed online. Everything seemed to install correctly. Opened rhino and it crashes if I run a script with `import compas`.

On this computer, I am running Windows 10, Rhino Version 7 SR26 (7.26.23009.7001, 2023-01-09), Python 3.8.16, conda 23.1.0.

mjnewsum, 2023-03-14:
When I setup my environment with conda, do I need to set the python version to 2.7 if I plan to use compas for Rhino 7? 

I am trying this now. After I activate my environment and try installing, it says it is installing to Rhino 6.0 and fails to find the IronPython folder. I tried to modify my install command to include v -7.0, and the error says 7.0 is an invalid choice. Only '5.0 or '6.0' are valid...

mjnewsum, 2023-03-14:
@gonzalocasas can I get some help with this install issue?

gonzalocasas, 2023-03-14:
On conda you would install python 3.x (3.7-3.10 have all been tested, 3.11 is a bit newer and some dependencies might be more difficult to install but in principle also works)

COMPAS knows how to deal with the 2.7 version that runs inside rhino, but your environment should contain an up to date, 3.x python.

I believe this usage of 2.7 on the conda side has triggered some of the problems you are seeing. I would first suggest you try to uninstall (`python -m compas_rhink.uninstall -v 7.0`) and the reinstall. If that still doesn't work, we can set up a quick zoom call to debug.

mjnewsum, 2023-03-14:
@gonzalocasas, thank you for getting back to me. 

I was only trying python 2.7 because I am unable to get python 3 versions to work. 
I tried installing python 3.10 and 3.8 before attempting 2.7. Everything has been uninstalled now ( as far as I can tell ). 

It would be great if we could hop on a call sometime to troubleshoot. I would really appreciate it!

mjnewsum, 2023-03-14:
When I uninstall, I get the following error:

Uninstalling COMPAS packages from Rhino 7.0 scripts folder:
C:\Users\mnewsum\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts

The following packages have been detected and will be uninstalled:

   compas_fab           OK
   compas_ghpython      OK
   roslibpy             OK
   compas_rhino         OK
   compas               OK
   compas_bootstrapper  OK

Running post-uninstallation steps...

   compas_fab           OK: Uninstalled 17 GH User Objects

One or more errors occurred:

   - AttributeError("'NoneType' object has no attribute 'version'")

Uninstall completed.

The user objects from compas do not get deleted. Only the compas_fab components are removed.
If I clean out this folder and install again, the compas user objects are not symlinked like the compas_fab components.

[image]

I don't know if this give any clues as to what is wrong with my installation.

brgcode, 2023-03-15:
do you have Rhino plugins like Ladybug installed by any chance?

tomvanmele, 2023-12-07:
@gonzalocasas is this resolved?

gonzalocasas, 2023-12-07:
Not sure. @mjnewsum did this work in the end?

jf---, 2024-04-08:
implicit communication: no follow up implies succes @gonzalocasas  || @tomvanmele  pls close

closed 2024-04-08 by tomvanmele

mjnewsum, 2024-04-12:
Sorry for missing the last few replies. I haven't looked back at this issue for a very long time.

I am no longer using the computer that had this issue, but it was never resolved. I would like to use Compas so I will try installing again on my new machine.

I will open a new ticket if I have any issues. Thank you for circling back to me.