compas-dev/compas #1140: Precision system

chenkasirer, 2023-05-03:
Related to: https://github.com/compas-dev/compas/pull/1057#discussion_r921978062

Some kind of a precision system can be useful which allows user to:
- get/set a globally desired unit (mm/meter/cm etc.)
- get/set a local version of the above
- get/set the desired tolerance for each

This would replace all sorts of local tolerance which aren't standardized and are defined per module/class/instance. 
Maybe in form of a `Precision` mixin which will by default fetch values from a global default and allow setting a modified local copy thereof.

gonzalocasas, 2023-05-03:
I always felt we needed to add built-in/native support for units across the board. Something like what `astropy` does: https://docs.astropy.org/en/stable/units/index.html

Because, well, it's important :) 

[image]

jf---, 2024-04-06:
@tomvanmele I think this can be closed with the Precision class being added in 2.0

yck011522, 2024-04-07:
@jf--- as far as I understand, the new Precision class solves the tolerance problem but did not really solve the lack of units problem that @gonzalocasas  mentioned.

jf---, 2024-04-07:
Gotcha, thanks for pointing out. But that seems another issue altogether: precision and metric(s) and 2 separate issues of concern.

gonzalocasas, 2024-04-08:
Closing this and opened #1319 for the units idea/discussion

closed 2024-04-08 by gonzalocasas