compas-dev/compas #1164: polyline animation: missing attribute

mo-nonym, 2023-07-07:
Hi. When trying to animate polylines with the bellow code, I get:

> AttributeError: 'PolylineArtist' object has no attribute 'width'

Snippet to reproduce the issue:

```python
from compas.geometry import Point, Polyline
from compas_plotters import Plotter

plot = Plotter(view=((-8.0, 16.0), (-5.0, 10.0)),
               figsize=(8.0, 5.0),
               dpi=100, bgcolor=(1.0, 1.0, 1.0),
               show_axes=False,
               zstack='zorder')
               
pl1 = Polyline([Point(0.000, 0.000, 0.000), Point(8.988, -4.384, 0.000)])
pl2 = Polyline([Point(0.000, 0.000, 0.000), Point(-8.660, -5.000, 0.000)])

polylines = [pl1, pl2]

def animate(plotter, objects, file):
    @plotter.on(interval=0.1,
                frames=len(objects),
                record=True,
                recording=file)
    def add_objects(frame):
        plotter.add(objects[frame])

animate(plot, polylines, "./demo.gif")

```

## Reference in source code:

https://github.com/compas-dev/compas/blob/5b171816f4d29334d2caf8ae6205705f6ccae5f4/src/compas_plotters/artists/polylineartist.py#L110C9-L110C49

## System info:

OS: Manjaro Linux x86_64
Kernel: 6.1.31-2-MANJARO
COMPAS: LooseVersion ('1.17.4')

chenkasirer, 2023-07-10:
Hi @mo-nonym! Thanks for opening this issue, and also tracking down the cause. 
Opened https://github.com/compas-dev/compas/pull/1166 with a fix.

closed 2023-07-10 by tomvanmele

mo-nonym, 2023-07-11:
Happy to help (:

PR #1166 (2023-07-10): fixed wrong attribute name in plotter artists
Fixed wrong attribute name used in `PolylineArtist` and `SegmentArtist` of `compas_plotter`.

Fixes #1164.

@tomvanmele should we remove `SegmentArtist`? I see it's not even registered for anything.

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [ ] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these  [... cut]