compas-dev/compas #1184: `Runtime error (SyntaxErrorException): unexpected token 'from'` when importing `compas`

chenkasirer, 2023-09-27:
**Describe the bug**
On some computers, when attempting to import `compas` inside GHPython, getting the following error:
```
Runtime error (SyntaxErrorException): unexpected token 'from'

Traceback:
  line 25, in <module>, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\data\encoders.py"
  line 11, in <module>, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\data\__init__.py"
  line 19, in <module>, "C:\Users\ckasirer\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\__init__.py"
  line 5, in script
```
This would typically happen when installing COMPAS to Rhino for the first time.

**To Reproduce**
Steps to reproduce the behavior:
1. Install COMPAS 
2. Install COMPAS to Rhino `python -m compas_rhino.install`
3. Open a new GHPython component
4. import `compas`

**Expected behavior**
`compas` should be imported successfully.

**Desktop (please complete the following information):**
 - OS: Mac/Win

**Additional context**
This has been spotted to happen where `numpy` is available in `sys.path` inside GH. This can be the case when, but not necessarily always, Ladybug is installed (since it downloads a python3 interpreter and adds its `site-packges` to `sys.path`).

closed 2023-09-27 by chenkasirer

Farzanbnh, 2023-09-29:
Hi chekasirer, I am getting the same error on a Microsoft Windows 11 Pro for Workstations

Runtime error (SyntaxErrorException): unexpected token 'from'

Traceback:
  line 26, in <module>, "C:\Users\offic\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\data\encoders.py"
  line 13, in <module>, "C:\Users\offic\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\data\validators.py"
  line 68, in <module>, "C:\Users\offic\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\data\__init__.py"
  line 45, in <module>, "C:\Users\offic\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\__init__.py"
  line 20, in <module>, "C:\Users\offic\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_ghpython\__init__.py"
  line 1, in script

gonzalocasas, 2023-09-29:
@Farzanbnh We release a new compas LTS version very very soon containing a fix for this. In the meantime, you could uninstall ladybug if you need a quick solution.

gonzalocasas, 2023-09-29:
@Farzanbnh we just released `compas 1.17.7`, it's already on pip, conda will still take some more hours but you could already try to update your environment with the pip release and that should solve the problem