compas-dev/compas #1192: jsonschema suddenly require Rust and Cargo installed

yck011522, 2023-10-04:
I have recently reinstalled compas usnig Conda.
However I have encountered a problem during installation.

```
Collecting referencing>=0.28.4 (from jsonschema->compas<2.0,>=1.14->compas-fab==0.28.0)
  Obtaining dependency information for referencing>=0.28.4 from https://files.pythonhosted.org/packages/be/8e/56d6f1e2d591f4d6cbcba446cac4a1b0dc4f584537e2071d9bcee8eeab6b/referencing-0.30.2-py3-none-any.whl.metadata
  Using cached referencing-0.30.2-py3-none-any.whl.metadata (2.6 kB)
Collecting rpds-py>=0.7.1 (from jsonschema->compas<2.0,>=1.14->compas-fab==0.28.0)
  Using cached rpds_py-0.10.3.tar.gz (17 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]

      Cargo, the Rust package manager, is not installed or is not on PATH.
      This package requires Rust and Cargo to compile extensions. Install it through
      the system's package manager or via https://rustup.rs/

      Checking for Rust toolchain....
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```
It appears to me that one of the dependency of compas, jsonschema got recently updated and the new build chain requires the Rust toolchain. This problem is also encountered by other software that uses jsonschema ([Example](https://github.com/python-poetry/poetry/issues/8160))I believe compas does not require the latest jsonschema and maybe possible to pin its versioning to jsonschema to >=4.17, <4.18. I have tested that the offending jsonschema version is 4.18

https://github.com/compas-dev/compas/blob/209201bb5e29618028b75588067c2b74c9df8829/requirements.txt#L5

tomvanmele, 2023-10-04:
@Licini could you make a quick PR for this?

Licini, 2023-10-04:
Made the PR @tomvanmele @yck011522 , I assume this is for the LTS version right?

tomvanmele, 2023-10-04:
i think for latest as well, since the problem will be universal...

gonzalocasas, 2023-10-04:
it's also ok to just do all these fixes for `main` always and I do the sync/backport to LTS, but if this is already based on lts, we can merge and release as is

gonzalocasas, 2023-10-04:
actually, I will do the opposite direction here, I will cherry pick into main

gonzalocasas, 2023-10-06:
backported to `main` https://github.com/compas-dev/compas/pull/1197

closed 2023-10-06 by gonzalocasas