compas-dev/compas #1195: Move `brep` pakcage out of `geometry`

chenkasirer, 2023-10-04:


jf---, 2024-04-08:
@tomvanmele -1 since surfaces and curves are there too. 🧐 a BRep is a topological layer on top of a surface, so strictly speaking brep could move to `compas.topology` but since the subpackages are broadly defines (`shapes`), I'd say hit close

[image]

tomvanmele, 2024-04-08:
breps were originally in `compas.brep` because that is where they are in `compas_occ`, and there this is the case because that is where they are in `OCC`, but after going back and forth a few times, we decided to leave them in geometry (at least for the time being)

closed 2024-04-08 by tomvanmele