compas-dev/compas #1211: compas2 and compas1 serialization

petrasvestartas, 2023-11-14:
**Describe the bug**
How compas1 .json files can be de-serialized from in compas2 alpha?

**To Reproduce**
E.g. repository .json files:
https://github.com/BlockResearchGroup/compas_assembly/tree/main/src/compas_assembly/data/samples

```python
from compas.data import json_load

if __name__ == "__main__":
    meshes = json_load('src/compas_assembly2/data_sets/crossvault.json')
```

**Expected behavior**
Working

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win
 - Python version 3.11.6
 - Python package manager conda

tomvanmele, 2023-11-14:
yes, i know. there is a slight change in the data serialisation format between COMPAS 1 and 2. i will add a conversion script...

xarthurx, 2024-09-06:
> yes, i know. there is a slight change in the data serialisation format between COMPAS 1 and 2. i will add a conversion script...

Is this implemented?