compas-dev/compas #1216: Better logging feedback

zaczhuozhang, 2023-11-29:
# Feature Request

As a user, I want the terminal logging to be more informational for easier debugging and tracing.

such output is hard to trace why:
```bash
File "C:\Users\zaczz\miniconda3\envs\compas-dev\lib\site-packages\compas\datastructures\halfedge\halfedge.py", line 439, in add_face
    if u not in self.halfedge[v]:
KeyError: 3
```

while developers can make this more informational by using logging tools like `loguru` and enumerating possible exceptions during the development.

also, I would suggest more rigorous type hint.

gonzalocasas, 2023-11-29:
Since we are talking about core here, I would not use `logging` or any derived tool, but instead I would add more assertions/checks and throw exceptions with meaningful messages

gonzalocasas, 2023-11-29:
Oh don't close it, please, just update it to request more assertions instead of logging

zaczhuozhang, 2023-11-29:
sure

jf---, 2024-04-08:
> by using logging tools like [loguru](https://loguru.readthedocs.io/en/stable/)

+1 loguru makes logging a joy