compas-dev/compas #1226: Line closes_point potential bug

petrasvestartas, 2023-12-15:
I think the line.closest_point methods gives bad results.

Example:

Given a line: `Line(Point(x=-6.0, y=0.0, z=0.0), Point(x=7.5, y=0.0, z=0.0))`
And a point:` Point(x=-5.0, y=0.0, z=0.0)`
The result is` 1.0`

But it should be:` 0.07407407407407407`

If this is an actual bug i suggest to change the current implementation to this one:

```python
def closest_point(line, point):
            rc = False
            t = 0.0

            if t:
                D = line.end - line.start
                DoD = D.length**2

                if DoD > 0.0:
                    if (point - line.start).length**2 <= (point - line.end).length**2:
                        t = D.dot(point - line.start) / DoD
                    else:
                        t = 1.0 + D.dot(point - line.end) / DoD

                    rc = True
                else:
                    t = 0.0
                    rc = True  # Closet point to a degenerate line works as well

            return rc, t
```

the original one is a bit simple, unless I am missing something:

```python
        a = self.start
        vector = point - a
        direction = self.direction
        t = vector.dot(direction)
        c = a + direction * t
        if return_parameter:
            return c, t
        return c
```

tomvanmele, 2023-12-15:
i would double check your test because the result seems correct to me...
```python
from compas.geometry import Point, Line
from compas_view2.app import App

line = Line(Point(-6.0, 0.0, 0.0), Point(7.5, 0.0, 0.0))
point = Point(-5, 0, 0)
closest, t = line.closest_point(point, True)

print(t)

viewer = App()
viewer.add(line)
viewer.add(closest)

viewer.run()
```
[image]

tomvanmele, 2023-12-15:
`t` is expressed wrt to the direction vector, which is a unit vector.
therefore `1.0` seems to make sense, since `Point(5, 0, 0)` is `Point(-6, 0, 0) + Vector(1, 0, 0) * 1.0`...

tomvanmele, 2023-12-15:
the difference with the C++ implementation lies mostly in the division by the squared length of the vector of the line, but that is implicitly included in the Python implementation because of the use of the unit direction vector.

petrasvestartas, 2023-12-16:
I thought that the `t` parameter can be directly used in `line.point_at(t)` method. 
This was the confusion.

Which means that in the current implementation it is `line.start+line.direction*t`

In this case it is not a bug, but intended behavior.

tomvanmele, 2023-12-17:
no, you're right. it should be possible to combine the two. it was not clear to me from your question/statements that this was the problem...

tomvanmele, 2023-12-17:
the domain in also defined as `0.0 => start` and `1.0 => end` so we should indeed update the implementation.

if don't know why the second "if" is necessary though. `t = v.dot(point - line.start) / vov` and `t = 1.0 + v.dot(point - line.end) / vov` seem equivalent to me...

```python
start = self.start
vector = point - start
direction = self.vector
t = vector.dot(direction) / self.length**2
closest = start + direction * t
if return_parameter:
    return closest, t
return closest
```

tomvanmele, 2023-12-17:
i will add a degeneracy check as part of the tolerance module update

tomvanmele, 2023-12-17:
if we don't care about the lookups, can be simplified to
```python
vector = point - self.start
t = vector.dot(self.vector) / self.length**2
closest = self.start + self.vector * t
if return_parameter:
    return closest, t
return closest
```

petrasvestartas, 2023-12-17:
`if don't know why the second "if" is necessary though. t = v.dot(point - line.start) / vov and t = 1.0 + v.dot(point - line.end) / vov seem equivalent to me...`

I checked different scenarios, I agree, there is no need to do that because the result is always the same.
The dot product projects one vector to the other, using the same starting point, so it is the same.
Also, there is never a need to know to which end the point is closer.

Thank you for changing to the normalized result: `t = vector.dot(self.vector) / self.length**2`
I think it is more clear when using normalization.

jf---, 2024-04-08:
@petrasvestartas || @petrasvestartas this seems resolved?

tomvanmele, 2024-04-08:
yes indeed

closed 2024-04-08 by tomvanmele