compas-dev/compas #1229: Plane.is_parallel wrong result as given in the documentation

petrasvestartas, 2023-12-18:
**Describe the bug**
In the documentation of the example code.
The result after running the method is False (latest compas2 from github.)

[image]

This method:
https://github.com/compas-dev/compas/blob/main/src/compas/geometry/plane.py#L354

I think, instead of this:
`return abs(self.normal.dot(other.normal)) == 1 - tol`
it has to be:
`1 - abs(self.frame.zaxis.dot(frame.zaxis)) < 1 - tol`

**To Reproduce**
```
    plane1 = Plane.worldXY()
    plane2 = Plane([1.0, 1.0, 1.0], [0.0, 0.0, 1.0])
    result = plane1.is_parallel(plane2)
    print(result)
```
prints False

**Expected behavior**
prints True

**Screenshots**
None

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] - Win
 - Python version [e.g. 2.7] - 3.11.6
 - Python package manager - conda

tomvanmele, 2023-12-18:
yes, the implementation is indeed wrong. will fix...
thanks!

tomvanmele, 2023-12-18:
will change to `abs(self.normal.dot(other.normal)) - 1 < tol`

jf---, 2024-04-08:
@tomvanmele I think you've addressed this in the `compas.tolerance` refactor?

tomvanmele, 2024-04-08:
yes indeed

closed 2024-04-08 by tomvanmele