compas-dev/compas #1241: API page display error

ZacZhangzhuo, 2023-12-27:
This `api` page: https://compas.dev/compas/latest/api.html is always wrong displayed foe me:

[image]

While this: https://compas.dev/compas/latest/api/index.html is fine.

tomvanmele, 2023-12-27:
@Licini this page is not meant to exist...

Licini, 2023-12-29:
> This `api` page: https://compas.dev/compas/latest/api.html is always wrong displayed foe me:
> 
> ![2023-12-12-27-11-16-22](https://private-user-images.githubusercontent.com/57562250/293011854-4d690ffd-d32d-4d12-822c-fca4c79db961.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDM4NTM5MjQsIm5iZiI6MTcwMzg1MzYyNCwicGF0aCI6Ii81NzU2MjI1MC8yOTMwMTE4NTQtNGQ2OTBmZmQtZDMyZC00ZDEyLTgyMmMtZmNhNGM3OWRiOTYxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyMzEyMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjI5VDEyNDAyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcyYjQyMWM3NzlkNjYyNDkwNWMxYThiODg5ZGE3NzY1OTgxMmQzYzNlMjY5NTY0YWY2YjU4N2Q1Mjc4MjY5NDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.cOOslVU8v84lPY_IyB0DXKy-4whj567BqYf84FxHlm8)
> 
> While this: https://compas.dev/compas/latest/api/index.html is fine.

This was a remained path of old website, it should not exist anymore. I've cleaned up and https://compas.dev/compas/latest/api/index.html is the only path from now.

closed 2023-12-29 by Licini