compas-dev/compas #1248: Vector operators

ZacZhangzhuo, 2024-01-04:
**Describe the bug**

for Vector peration `*`, like:
```python
>>> u = Vector(1.000, 1.000, 0.000)
>>> u * 2
Vector(2.000, 2.000, 0.000)
```
if we do:
`Vector * float`: It is fine.
`float * Vector`: TypeError: unsupported operand type(s) for *: 'float' and 'Vector'.

**Expected behavior**
I think this could be fixed with minimal effort just for a better user experience. :)

gonzalocasas, 2024-01-04:
Can it really be solved easily? I have the feeling it's not possible to define the binding of the star operator to take right hand operand as precedence...

tomvanmele, 2024-01-04:
yes, we could implement the "r" versions of the magic methods, but i don't know if there would be any unwanted side-effects...

```python
def __rmul__(self, factor):
    ...
```

gonzalocasas, 2024-01-04:
Ah true 👍

ZacZhangzhuo, 2024-01-05:
yes, it's not a big issue. mathematically the floating number is preferred in the front. like `a = k b`, a, b are vectors.

jf---, 2024-04-08:
@tomvanmele you're right that __rmul__ would address the issue, and @ZacZhangzhuo is right in that operators in Vectors should be complete -- but `float * Vector` is quite counter intuitive?

tomvanmele, 2024-04-08:
i agree, that is why never implemented the "r" versions. i think it is easy enough for the user to be explicit about their intent by writing the compas object first...

jf---, 2024-04-08:
@tomvanmele I know I'm being frantic, but seems like this issue can be closed? The broader concept of vec ops isn't discussed.

closed 2024-04-09 by tomvanmele

xarthurx, 2024-04-29:
> @tomvanmele you're right that **rmul** would address the issue, and @ZacZhangzhuo is right in that operators in Vectors should be complete -- but `float * Vector` is quite counter intuitive?

It is actually quite common and intuitive, especially when you compute a "WEIGHTED * long-expression"....