compas-dev/compas #1269: ImportError: cannot import name 'cross' from 'scipy' in COMPAS LTS

chenkasirer, 2024-01-22:
reported by @mlstich (thanks!)

Getting a `ImportError: cannot import name 'cross' from 'scipy'` with the recently released `scipy=1.12.0`.
Since we got rid of `numerical` in 2.0.0 this possibly only effects the LTS. 

Should we just pin `scipy`?

**To Reproduce**
1. `conda install compas=1.17.9 scipy=1.12.0`
2. `from compas.geometry import Box`

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\compas\geometry\__init__.py", line 827, in <module>
    from .bbox import bounding_box, bounding_box_xy  # noqa: E402
  File "C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\compas\geometry\bbox\__init__.py", line 10, in <module>
    from .bbox_numpy import *  # noqa: F401 F403
  File "C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\compas\geometry\bbox\bbox_numpy.py", line 26, in <module>
    from compas.numerical import pca_numpy
  File "C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\compas\numerical\__init__.py", line 137, in <module>
    from .operators import grad, div, curl
  File "C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\compas\numerical\operators.py", line 10, in <module>
    from scipy import cross
ImportError: cannot import name 'cross' from 'scipy' (C:\Users\ckasirer\anaconda3\envs\scipy\lib\site-packages\scipy\__init__.py)
```
**Desktop (please complete the following information):**
 - OS: Windows10
 - Python 3.10
 - conda
 - `scipy==1.12.0`

tomvanmele, 2024-01-22:
get cross from numpy instead

jackbow, 2024-01-22:
For those looking for an immediate fix for their own use of compas, I added scipy to my conda environment and pinned it to version 1.11.4 and everything works again.

gonzalocasas, 2024-01-22:
@jackbow we just released a version with this fix (`1.17.10`), it's already available on pypi, it will reach conda in a few hours

gonzalocasas, 2024-01-24:
@chenkasirer we can close this, right?

chenkasirer, 2024-01-24:
yes. thanks!

closed 2024-01-24 by chenkasirer